Joshua-Riek/ubuntu-rockchip #319: Orange Pi 5B Bluetooth audio issue

Aashitech, 2023-08-19:
On Opi-5B (having wifi/BT on board) the BT connected speakers audio is not smooth and distorted , did some one else too have observed this thing  ? using v1.22 image.

PS : Joshua would you like to extend your work on RK3566 based SBC also ? if yes then please share your details as I can get you a latest Orange Pi 3B ( RK3566) shipped directly from China . As there is no way to have private messages on Github so I am offering this here only .

Joshua-Riek, 2023-08-19:
I have tested Bluetooth audio with my JBL FLIP 5 speaker, and the sound is clear. Make sure the antennas are secure and attached. It could have a problem with your specific Bluetooth speaker, but I do not know.

I do not plan to add support for the Orange Pi 3B right now, but maybe in the future.

Aashitech, 2023-08-19:
Ok , I will try with some other set of BT speakers and will report back here .
For Opi-3B  (RK3566) if you wish and decides to add support then please let me know ,and in meantime (if you wish) then we can ship you a OPI-3B .

Joshua-Riek, 2023-08-19:
I will keep that in mind, it's appreciated. Right now I need to focus on build system improvements, and I would like to work on mainline Linux support for the Orange Pi 5.

Aashitech, 2023-08-19:
update : going to settings /sound/ and switching output "configuration " from "A2DP sink" to something else and then back to "A2DP sink" again resolve this distorted audio issue , very weird though   ....

Joshua-Riek, 2023-08-22:
That is strange, and I'm glad you got it working. But I don't think this is something that I can fix.

closed 2023-08-26 by Joshua-Riek