Joshua-Riek/ubuntu-rockchip #320: Orange Pi 5 network manager reset configuration after reboot

1984kav, 2023-08-19:
Orange Pi 5 Ubuntu desktop network manager reset configuration after reboot

Joshua-Riek, 2023-08-19:
I need some more information on how to reproduce this issue and what settings are being reset.

1984kav, 2023-08-20:
I set up an Ethernet network connection profile - I specify static IP and MAC addresses.  After a reboot, the profile settings are reset.  This prevents me from fixing the IP address.  Thanks for your work.

1984kav, 2023-08-21:
> 

I set up an Ethernet network connection profile - I specify static IP and MAC addresses. After a reboot, the profile settings are reset. This prevents me from fixing the IP address. Thanks for your work.

1984kav, 2023-08-28:
On OrangePi 5 plus Ubuntu Desktop this bug does not exist.

1984kav, 2023-08-30:
Problem solved by reinstalling the system

closed 2023-08-30 by Joshua-Riek