Joshua-Riek/ubuntu-rockchip #324: Bluetooth device is not found

s1n7ax, 2023-08-20:
I have a Sony WH1000XM4 headphone and a Sound Core 2.  I'm running Orange Pi 5 Plus with Wifi and Bluetooth module. I have tried few Orange PI official images. In some some of them Bluetooth devices are not detected but on Orange Pi Arch Linux it works (Though wifi works on all of them).

Module name on Ali Express:

```
Orange Pi 5 Plus WiFi 6 + BT 5.2 Wireless Module RTL8852BE M.2 E Key Interface 1800M Dual Brand 2.4G 5G
```

Steps to reproduce:
1. run bluetoothctl
2. run default-agent
3. run power on
4. run scan on
5. turn on pairing mode in the headphone
6. wait until headphone is detected

Expected to find one or more devices but non of the devices are detected.

```
> scan on
[CHG] controller XX:XX:XX:XX:XX:XX Discovering: yes

^^^ once a device is detected, it should show the id here but not found
```

Joshua-Riek, 2023-08-20:
That WiFi + BT module has been giving me much trouble recently. 

Please add the below lines to `/etc/modprobe.d/blacklist.conf` and then reboot your system. Hopefully, this will work.
```
blacklist btusb
blacklist btrtl
blacklist btbcm
blacklist btintel
```

lukaszsobala, 2023-08-20:
@Joshua-Riek I have a similar problem with ax210 from Intel. On kernel 5.10.160-10 it worked but I think after this update 5.10.10.160-11) it stopped . I tried blacklisting `btintel` or forcing it to load in modprobe - nothing works.

I have the latest firmware from Intel downloaded and put in the correct directory.

lukaszsobala, 2023-08-20:
@Joshua-Riek I have a similar problem with ax210 from Intel. On kernel 5.10.160-10 it worked but I think after this update 5.10.10.160-11) it stopped working. I tried blacklisting `btintel` or forcing it to load in modprobe - nothing works.

I have the latest firmware from Intel downloaded and put in the correct directory.

Joshua-Riek, 2023-08-20:
> @Joshua-Riek I have a similar problem with ax210 from Intel. On kernel 5.10.160-10 it worked but I think after this update 5.10.10.160-11) it stopped . I tried blacklisting `btintel` or forcing it [...]

Can you put dmesg into a text file and attach it here? If the Bluetooth firmware for ax210 is unable to load there will be an error message.

lukaszsobala, 2023-08-20:
hello,

here is the output of `sudo dmesg|grep -i blue`:
```
[    7.512465] Bluetooth: Core ver 2.22
[    7.512481] Bluetooth: HCI device and connection manager initialized
[    7.512485] Bluetooth: HCI socket layer initialized
[    7.512488] Bluetooth: L2CAP socket layer initialized
[    7.512494] Bluetooth: SCO socket layer initialized
[   20.215307] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   20.215322] Bluetooth: BNEP filters: protocol multicast
[   20.215341] Bluetooth: BNEP socket layer initialized

```
and here is the full dmesg.
[5a_dmesg.txt](https://github.com/Joshua-Riek/ubuntu-rockchip/files/12387759/5a_dmesg.txt)

The bluetooth controller does show up in `lsusb` but does not in `rfkill`. Before the kernel update it did (but I had the same problem for some time, not sure what fixed it).

Joshua-Riek, 2023-08-20:
@lukaszsobala I tested this out with my ax210 and the 5.10.160-11 kernel, and both WiFi and Bluetooth are working fine, I don't even need to backlist any modules. From your logs, the firmware is loaded correctly, and there should be no issues.

lukaszsobala, 2023-08-20:
So why does it not work? :(

Joshua-Riek, 2023-08-22:
Im not sure, the firmware and drivers are working fine. My guess is something on your end.

Deiiji, 2023-08-25:
Hello, I have same issue with this realtek card, is same in all distros, WiFi work but no BT, module is also cold, very cold.

Joshua-Riek, 2023-08-25:
There were two wifi and bt modules mentioned here, which are you referring to? From my end, both work fine on multiple devices.

s1n7ax, 2023-08-25:
Unfortunately I switched to debian server right after this issue. There is no way I can validate the fix. You can close this if you this other problems mentioned here are unrelated.

Deiiji, 2023-08-26:
Hello again, first, thank you for great work..
After blacklist bt modules bt work but no more WiFi., module 8852

Joshua-Riek, 2023-08-26:
Open an issue specifically for wifi or Bluetooth for the rtl8852be. I'm closing this issue.

closed 2023-08-26 by Joshua-Riek