Joshua-Riek/ubuntu-rockchip #325: [Orange Pi 5] Chromium error after "sudo  apt dist-upgrade"

saulgodino, 2023-08-21:
Hi, I did the execution of the commands bellow in the order:
1) sudo apt update
2) sudo apt upgrade
3) sudo apt dist-upgrade
Then the chromium browser was updated to 1:114.0.5735.0-0ubuntu1~ppa2~22.04.1+rkmpp1 version. But some pages are no longer opening anymore. Showing the error bellow. Example: globo.com
"Aw, Snap!
Something went wrong while displaying this webpage
Error code: SIGTRAP"

Joshua-Riek, 2023-08-21:
@amazingfate updated chromium to version 114 on his ppa, and maybe there is some issue with the new version? I do not know much about chromium.

Joshua-Riek, 2023-08-21:
I know this build works for sure. 

https://launchpad.net/~liujianfeng1994/+archive/ubuntu/rockchip-multimedia/+build/25689009

You can try to install the old one and hold the packages to prevent an update.
```bash
wget https://launchpadlibrarian.net/657101666/chromium-browser-l10n_110.0.5481.4-0ubuntu1~jammy~rkmpp20230130+1_all.deb
wget https://launchpadlibrarian.net/657101668/chromium-browser_110.0.5481.4-0ubuntu1~jammy~rkmpp20230130+1_arm64.deb
wget https://launchpadlibrarian.net/657101670/chromium-chromedriver_110.0.5481.4-0ubuntu1~jammy~rkmpp20230130+1_arm64.deb
wget https://launchpadlibrarian.net/657101671/chromium-codecs-ffmpeg-extra_110.0.5481.4-0ubuntu1~jammy~rkmpp20230130+1_arm64.deb
wget https://launchpadlibrarian.net/657101673/chromium-codecs-ffmpeg_110.0.5481.4-0ubuntu1~jammy~rkmpp20230130+1_arm64.deb
sudo dpkg -i *.deb
sudo apt-mark hold chromium-browser chromium-browser-l10n chromium-chromedriver chromium-codecs-ffmpeg chromium-codecs-ffmpeg-extra
```

amazingfate, 2023-08-21:
I can open `https://globo.com`.

Joshua-Riek, 2023-08-21:
I tested version 114 of chromium on my image and monkaBlyat's and received the same error. Maybe there is some issue.

saulgodino, 2023-08-22:
> ```shell
> sudo apt-mark hold chromium-browser chromium-browser-l10n chromium-chromedriver chromium-codecs-ffmpeg chromium-codecs-ffmpeg-extra
> ```

It worked for me. Reinstalled previous version. Thanks.

amazingfate, 2023-08-24:
Fixed in new version `1:114.0.5735.35-0ubuntu1~ppa2~22.04.1+rkmpp`.

sdeleeuw, 2023-08-25:
@amazingfate 

I noticed that upgrading `chromium-browser` requires `chromium-browser-l10n` to be removed. Because of this, only a dist-upgrade will actually upgrade `chromium-browser`. Is the l10n package intentionally removed from the build?

```
sander@orangepi5:~$ apt --simulate upgrade
NOTE: This is only a simulation!
      apt needs root privileges for real execution.
      Keep also in mind that locking is deactivated,
      so don't depend on the relevance to the real current situation!
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
Get more security updates through Ubuntu Pro with 'esm-apps' enabled:
  libopenexr25 libmagickcore-6.q16-6-extra libmagickwand-6.q16-6
  libmagickcore-6.q16-6 imagemagick-6-common
Learn more about Ubuntu Pro at https://ubuntu.com/pro
The following packages have been kept back:
  chromium-browser chromium-codecs-ffmpeg-extra
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.

sander@orangepi5:~$ apt --simulate dist-upgrade
NOTE: This is only a simulation!
      apt needs root privileges for real execution.
      Keep also in mind that locking is deactivated,
      so don't depend on the relevance to the real current situation!
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Calculating upgrade... Done
Get more security updates through Ubuntu Pro with 'esm-apps' enabled:
  libopenexr25 libmagickcore-6.q16-6-extra libmagickwand-6.q16-6
  libmagickcore-6.q16-6 imagemagick-6-common
Learn more about Ubuntu Pro at https://ubuntu.com/pro
The following packages will be REMOVED:
  chromium-browser-l10n
The following packages will be upgraded:
  chromium-browser chromium-codecs-ffmpeg-extra
2 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
Remv chromium-browser-l10n [1:110.0.5481.4-0ubuntu1~jammy~rkmpp20230130+1]
Inst chromium-browser [1:110.0.5481.4-0ubuntu1~jammy~rkmpp20230130+1] (1:114.0.5735.35-0ubuntu1~ppa2~22.04.1+rkmpp rockchip legacy multimedia packages:22.04/jammy [arm64]) []
Inst chromium-codecs-ffmpeg-extra [1:110.0.5481.4-0ubuntu1~jammy~rkmpp20230130+1] (1:114.0.5735.35-0ubuntu1~ppa2~22.04.1+rkmpp rockchip legacy multimedia packages:22.04/jammy [arm64])
Conf chromium-browser (1:114.0.5735.35-0ubuntu1~ppa2~22.04.1+rkmpp rockchip legacy multimedia packages:22.04/jammy [arm64])
Conf chromium-codecs-ffmpeg-extra (1:114.0.5735.35-0ubuntu1~ppa2~22.04.1+rkmpp rockchip legacy multimedia packages:22.04/jammy [arm64])
```

amazingfate, 2023-08-26:
Debs of architecture all is build by armhf on launchpad, which is failed at the moment.

sdeleeuw, 2023-08-28:
@amazingfate @Joshua-Riek 

The usual `apt upgrade` now updated `chromium-browser` to the latest version.
An updated `chromium-browser-l10n` is also available from the `rockchip-multimedia` PPA.

I also visited "globo.com" and it worked fine. Guess this issue can be closed now.

Thanks.

Joshua-Riek, 2023-08-29:
I'm glad to hear it's working. Thanks again for the patch, amazingfate.

closed 2023-08-29 by Joshua-Riek