Joshua-Riek/ubuntu-rockchip #326: Error while sudo apt update on Orange Pi 5

lreflectl, 2023-08-21:
When trying to update system packages with 'sudo apt update', I get following messages:

```shell
Failed to fetch https://ppa.launchpadcontent.net/jjriek/panfork-mesa/ubuntu/dists/jammy/InRelease  403  Forbidden [IP: 185.125.190.52 443]
The repository 'https://ppa.launchpadcontent.net/jjriek/panfork-mesa/ubuntu jammy InRelease' is no longer signed.
Updating from such a repository can't be done securely, and is therefore disabled by default.
``` 

Full log:

```shell
reflect@opi5:~$ sudo apt update
Hit:1 http://ports.ubuntu.com/ubuntu-ports jammy InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports jammy-updates InRelease             
Hit:3 http://ports.ubuntu.com/ubuntu-ports jammy-backports InRelease           
Hit:4 http://ports.ubuntu.com/ubuntu-ports jammy-security InRelease            
Hit:5 https://download.docker.com/linux/ubuntu jammy InRelease                             
Hit:6 http://packages.microsoft.com/repos/code stable InRelease                            
Hit:7 https://ppa.launchpadcontent.net/danielrichter2007/grub-customizer/ubuntu jammy InRelease
Hit:8 https://ppa.launchpadcontent.net/jjriek/orangepi5/ubuntu jammy InRelease
Err:9 https://ppa.launchpadcontent.net/jjriek/panfork-mesa/ubuntu jammy InRelease
  403  Forbidden [IP: 185.125.190.52 443]
Err:10 https://ppa.launchpadcontent.net/jjriek/rockchip-multimedia/ubuntu jammy InRelease
  403  Forbidden [IP: 185.125.190.52 443]
Reading package lists... Done                              
E: Failed to fetch https://ppa.launchpadcontent.net/jjriek/panfork-mesa/ubuntu/dists/jammy/InRelease  403  Forbidden [IP: 185.125.190.52 443]
E: The repository 'https://ppa.launchpadcontent.net/jjriek/panfork-mesa/ubuntu jammy InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Failed to fetch https://ppa.launchpadcontent.net/jjriek/rockchip-multimedia/ubuntu/dists/jammy/InRelease  403  Forbidden [IP: 185.125.190.52 443]
E: The repository 'https://ppa.launchpadcontent.net/jjriek/rockchip-multimedia/ubuntu jammy InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

Looks like expired certificates somewhere. About a month ago the command has been working fine.

Joshua-Riek, 2023-08-21:
It looks like you are using an extremely old build. Some of the ppa no longer exist or are maintained. I'm sorry for the inconvenience but please download the latest release and re-install the OS.

lreflectl, 2023-08-21:
Thank you for your response. I'll install fresh build, just thought that all distro updates can be installed from a command line.

Joshua-Riek, 2023-08-21:
Yes, most updates can be installed through the command line, but there are a few exceptions. Also, I made a few mistakes in the early days of this project when it came to the package repository that I could not revert. I'm still learning myself and was not expecting so many users in such a short time span.