Joshua-Riek/ubuntu-rockchip #327: Request to add 2 device support in flash-kernel

erd-yunmu, 2023-08-22:
I enabled Ubuntu on 2 devices using your build script, but it crashes immediately after entering the installer settings.
https://github.com/Joshua-Riek/ubuntu-rockchip/issues/275 ; Seeing that there are similar problems related to me, so I am here to apply to you to add 2 devices to solve the problem of the crash of the installer, thank you!
```
	model = "Embedfire LubanCat-4";
	compatible = "rockchip,rk3588s-lubancat-4", "rockchip,rk3588";
```
```
	model = "Embedfire LubanCat-5";
	compatible = "rockchip,rk3588-lubancat-5", "rockchip,rk3588";
```

Joshua-Riek, 2023-08-22:
Sure, I can add them to the flash kernel package it's an easy change. I will do this after I get home from work today and let you know when I upload the change.

erd-yunmu, 2023-08-23:
ok，thank you！

Joshua-Riek, 2023-08-23:
The package is building now and should be uploaded in 1 or 2 hours.

https://launchpad.net/~jjriek/+archive/ubuntu/rockchip/+build/26530744

erd-yunmu, 2023-08-23:
Test issue solved!
thank you for your help！

closed 2023-08-23 by erd-yunmu