Joshua-Riek/ubuntu-rockchip #329: Orange Pi 5 plus crashes while watching youtube.

AndriusZimmer, 2023-08-22:
Hi, I'm a newbie and I'm trying to learn. I have researched similar errors and found information about PAN_MESA_DEBUG=gofaster and bad power supplies. I don't think 'gofaster' is the issue in my case because I installed the latest version (v1.23). My power supply is 5V 4.5A. It's not the original one, but I believe it's working fine. I would buy another one for testing, but I have to buy from another country since my country's prices are overpriced, and it's hard to find. I am posting the lo. I am afraid that it's a problem with my Orange Pi. Thank you for your time.

[log.zip](https://github.com/Joshua-Riek/ubuntu-rockchip/files/12413592/log.zip)

Additional info:
It's an Orange Pi 5 Plus 8GB. It has an AX210NGW Wi-Fi board and a Netac 256GB NVMe. I use a USB receiver for both the mouse and keyboard. I have already run a stress test for 1 hour with HardInfo and encountered no issues.

Joshua-Riek, 2023-08-30:
I would recommend another 5v 4a power supply, the system is quite stable and the common cause of crashing is related to power.

closed 2023-08-30 by Joshua-Riek

AndriusZimmer, 2024-03-08:
Hello, I bought a new power supply, and the system is working flawlessly. Thanks for the reply, you were right. Thanks for your support and for the incredible project.