Joshua-Riek/ubuntu-rockchip #332: chroot: failed to run command ‘/bin/true’: No such file or directory

MrLiujiefei, 2023-08-24:
When i run sudo ./build.sh -do -l --board=rock-5b on ubuntu 22.04 system,An error has occurred：

2023-08-24 23:14:24 URL:http://ports.ubuntu.com/ubuntu-ports/pool/main/z/zlib/zlib1g_1.2.11.dfsg-2ubuntu9_arm64.deb [56848/56848] -> "/home/ljf/RK3588/Linux_SDK/ubuntu-rockchip/build/rootfs//var/cache/apt/archives/partial/zlib1g_1%3a1.2.11.dfsg-2ubuntu9_arm64.deb" [1]
chroot: failed to run command ‘/bin/true’: No such file or directory

how can i do

Joshua-Riek, 2023-08-24:
There is a build log in the folder build/log/, please upload the file here to GitHub. I need to know what line this error occurred.

Joshua-Riek, 2023-08-24:
I tested a build with the command you ran and encountered no issues.
[build-20230824144329.log](https://github.com/Joshua-Riek/ubuntu-rockchip/files/12434093/build-20230824144329.log)

MrLiujiefei, 2023-08-25:
[build-20230824230503.log](https://github.com/Joshua-Riek/ubuntu-rockchip/files/12434378/build-20230824230503.log)
[debootstrap.log](https://github.com/Joshua-Riek/ubuntu-rockchip/files/12434379/debootstrap.log)
This is my build logs

MrLiujiefei, 2023-08-25:
i don't download kernel and only build rootfs

Joshua-Riek, 2023-08-25:
I'm not sure why you get that error. I would try on a new Ubuntu 22.04 install, then make sure to download all dependencies.

closed 2023-08-30 by Joshua-Riek

oicu, 2024-09-07:
> When i run sudo ./build.sh -do -l --board=rock-5b on ubuntu 22.04 system,An error has occurred：
>
> 2023-08-24 23:14:24 [...]

```
ldd rootfs/bin/true
```

```
sudo cp /lib/x86_64-linux-gnu/libc.so.6 rootfs/usr/lib/x86_64-linux-gnu/
sudo cp /lib64/ld-linux-x86-64.so.2 rootfs/usr/lib64/
```