Joshua-Riek/ubuntu-rockchip #336: Kodi: green screen when opening app

Singular112, 2023-08-29:
Hi.
I have orange pi 5.
On v123, v122 (maybe, and other) versions of ubuntu, when I'm opening Kodi, installed from repo (apt install kodi), the green screen with an artefact appears.

What information do you need to fix the issue ?

Joshua-Riek, 2023-08-29:
I do not modify Kodi. That is out of the scope of this project. I know it has always had problems, but you need to launch Kodi from the login screen by selecting Kodi in the bottom right corner.

branislavhesko, 2023-09-04:
Hello, I found a simple workaround:

Try to navigate into KODI settings, select windowed mode, green artefact should disappear. Afterwards switch again to fullscreen and KODI should be as expected. This artefact is supposedly caused by some Wayland issues.

closed 2025-02-17 by github-actions[bot]