Joshua-Riek/ubuntu-rockchip #338: rtl8852be WIFI/bluetooth issues - how to reproduce the issue

arunas-smala, 2023-08-29:
I know this WiFi/BT module sucks, just letting you know how to reproduce the issues.

After fresh install WiFi works perfectly fine. Bluetooth adapter found, but no bt devices on the list.

After editing `blacklist.conf` as you advised in other comments - reboot and both bluetooth and wifi works! Reboot - still works. Seems fixed.

Shutdown, unplug power (not sure if needed), boot again - WiFi no networks found... Reboot - the same..

How to "fix" it? Disable bluetooth, shutdown, boot again - everything works: both, wifi and bt.

Long story short, after blacklisting - both wifi and bt works, just you have to "disable" bluetooth before shutdown. Otherwise next boot wifi will not work (will not find any networks). Seems like if both wifi and bt enabled - it causes some conflicts during boot time. But if only wifi enabled - you can enable bt later and both will work.. strange.

I'm currently on OrangePi os (arch), just because of this issue.. But I like your OS much more, would come back if this issue could be solved.. Thank you for your work!

arunas-smala, 2023-08-30:
P.S. I would buy another wifi/bt unit if someone can advice which one. On aliexpress I see rtl8852be as the only option for opi 5 plus, and not sure if any m.2 module would work, because I see one unit for Orange pi 5, and it says "will NOT work with orange pi 5 PLUS"..

rusudanion, 2023-09-02:
hello 

> P.S. I would buy another wifi/bt unit if someone can advice which one. On aliexpress I see rtl8852be as the only option for opi 5 plus, and not sure if any m.2 module would work, because I see one unit for Orange pi 5, and it says "will NOT work with orange pi 5 PLUS"..

hello i use ax210 ,works good.

PeterGamma, 2023-09-09:
Hi,

biblioman09 made a review on his YouTube channel with an Orange Pi Plus and a PCIe WIFI6 + Bluetooth:

https://youtu.be/qxig20UcPpw

He did not use the official Bluetooth module from Orange, but a different sold on Aliexpress:

https://www.aliexpress.com/item/1005005867630310.html?aff_fcid=831d1b3c922646fc8369371b66514f12-1694274924231-01206-_Dkoc32n&tt=CPS_NORMAL&aff_fsk=_Dkoc32n&aff_platform=shareComponent-detail&sk=_Dkoc32n&aff_trace_key=831d1b3c922646fc8369371b66514f12-1694274924231-01206-_Dkoc32n&terminal_id=64a25cf22e7b461cae1b2cde5b5e78c5&afSmartRedirect=y

In the description of the module it says it can used on 

- Samsung/Dell/Sony/ACER/ISUS/MSI/ Clevo/Terransforce/Hasee etc.,
- but it does not work on IBM/Thinkpad/lenovo/hp laptop

I have listed on my site what he got to work with the module and the  Orange Pi Plus:

https://petergamma.org/demonstration-which-shows-the-function-of-the-orange-pi-5-plus-with-a-wifi-bluetooth-module-in-ubuntu-but-still-no-proof-of-function-of-bluetooth-keyboard-mouse/

Bluetooth and WIFI seems still to be an issue with the Orange Pi 5 (Plus) and it is highly desirable 
that it be solved.

Joshua-Riek, 2023-09-09:
I tested two Orange Pi 5 Plus devices and rtl8852be WiFi + Bluetooth cards. I can not reproduce this issue, modifying `blacklist.conf` also is not required, the module will work out of the box.

closed 2023-09-09 by Joshua-Riek