Joshua-Riek/ubuntu-rockchip #339: Orange Pi 5b: cannot change resolution from terminal

raquibahmed, 2023-08-29:
I have 2 monitors at home, one is MSI ultra wide and another aftermarket 8 inch (no brand). If I connect Orange Pi 5b, to the MSI monitor no issues, I have display and everything works as expected but when I connect to the aftermarket no brand small monitor there is no display. I am on 1.23 release.

To test the monitor, I have flashed a SD card with Armbian Lunar Lobster image from the nightly build, and same I had no display. With ssh access, I was able to use `xrandr -s 800x600`, which changed the resolution and the monitor had display. After that I could access Display settings and play around no problem. So the aftermarket display works.

So I did the same with Ubuntu 1.23 release, gained ssh access and `xrandr` shows me `XWAYLAND0` is connected. `xrandr` command no longer works to change resolution and nothing seems to work to change the resolution via terminal. I come here as the last resort to try to change the resolution via terminal, so appreciate your help on this. Here is the output from the `xrandr`. Kindly let me know if you require any more outputs and I can get them.

```
opi5b@opi5b-desktop:~$ export DISPLAY=:0.0
opi5b@opi5b-desktop:~$ xrandr
Screen 0: minimum 16 x 16, current 1024 x 600, maximum 32767 x 32767
XWAYLAND0 connected primary 1024x600+0+0 (normal left inverted right x axis y axis) 470mm x 260mm
   1024x600      59.55*+
   800x600       59.47  
   640x480       59.38  
   320x240       59.52  
   720x480       59.71  
   640x400       59.20  
   320x200       58.96  
   1024x576      59.58  
   864x486       59.45  
   720x400       59.55  
   640x350       58.91  
```

Thanks again

Joshua-Riek, 2023-08-29:
Armbian uses the x11 display manager by default, which works with xrandr. This distro defaults to wayland as it performs better and is much more stable. However, xrandr may not act right when using wayland as the tool is built for x11.

raquibahmed, 2023-08-29:
Thank you for the quick response. Is there any command to change the resolution for wayland from terminal?

Joshua-Riek, 2023-08-29:
I have no idea. You can try a Google search.

closed 2023-08-30 by Joshua-Riek