Joshua-Riek/ubuntu-rockchip #343: chmod +x .run file

zer0s1x, 2023-09-02:
Orange PI 5 user here.

I'm trying to .run a new fydeos v17 file in the terminal, obviously using the "chmod +x filename.run" command. It's supposed to just generate an .img file from it so i can flash it in my sd card. The problem is, when i run this command, it opens the *supported boards* thing showing options for all the 3 orange pi 5 boards, but within these options there's no sd card alternative, only nvme and sata. It's quite dumb to have only these options, i just want to generate my image to boot to my sd card and that's it. Isn't there another command to .run the file or skip this thing and generate the image itself right away?

Joshua-Riek, 2023-09-02:
I don't know how Fydeos works or how to install it. You should open an issue on their GitHub repo.

closed 2023-09-02 by Joshua-Riek

zer0s1x, 2023-09-02:
True. I tested other .run files and looks like they don't have the same problem. Thought this was a problem with this distro in specific, but looks like it's a fydeos thing. Sorry and thanks for the suggestion.