Joshua-Riek/ubuntu-rockchip #345: 4K video with  h265 encoding can't play properly

felix2ch, 2023-09-03:
Test for two movies:
1. Avatar.The.Way.of.Water.2022.2160p.UHD.Bluray.REMUX.HDR10.HEVC.TrueHD.Atmos.7.1-GHD[TGx]
Play blocked when drag progress bar.

2. Billy.Lynns.Long.Halftime.Walk.2016.2160p.HFR.UHD.BluRay.HDR.DTS-HD.MA5.1.x265-ULTRAHDCLUB
Play failed totally. 

Maybe HDR not supported?
1080p h265 movies play fine.

Joshua-Riek, 2023-09-03:
The Wayland display manager does not support HDR video playback. The alternative is to use the X11 display manager, but it has very bad performance and a lot of graphical glitches.

nguyenhunga5, 2023-09-26:
You can use ffplay to play this file.

nguyenhunga5, 2023-12-08:
@Joshua-Riek I saw someone in here can play h265 https://forum.armbian.com/topic/25957-guide-kodi-on-orange-pi-5-with-gpu-hardware-acceleration-and-hdmi-audio/?do=findComment&comment=174386
Could you take a look?

closed 2025-02-17 by github-actions[bot]