Joshua-Riek/ubuntu-rockchip #349: WiFi AX210 not working. Only bluetooth.

c0h1b4, 2023-09-06:
I know it is a repeated problem, but I just got my Orange PI 5 plus today, installed the Intel AX210 and I have absolutely no WiFi. Only Bluetooth.
The error is the same as a lot of people described before, but the kernel is not installing/running the correct wifi driver.

I can see the messages (dmesg): 

==========================================
=== Launching Wi-Fi driver! (Powered by Rockchip) ===
==========================================
Realtek 8821CS SDIO WiFi driver (Powered by Rockchip,Ver 5.14.2-18-g6011b0372.20220328) init.
[WLAN-RFKILL]: rockchip_wifi_power: 1
[WLAN-RFKILL]: rockchip_wifi_power: rfkill-wlan driver has not Successful initialized
...
[WLAN-RFKILL]: can't find rockchip,grf property
etc.

I don't see any data about the AX210 on the dmesg.

Tried all the instructions from other people here but had no success.

c0h1b4, 2023-09-06:
I'm using the 1.24 version

Joshua-Riek, 2023-09-06:
The dmesg log is normal and it should be working with no problem, please try the below.

Try to add the below lines to `/etc/modprobe.d/blacklist.conf` and then reboot your system. 
```
blacklist btusb
blacklist btrtl
blacklist btbcm
blacklist btintel
```

c0h1b4, 2023-09-06:
Now I have no Bluetooth and no WiFi.

Joshua-Riek, 2023-09-06:
Provide full dmesg logs please in a text file.

c0h1b4, 2023-09-06:
[dmesg.zip](https://github.com/Joshua-Riek/ubuntu-rockchip/files/12543054/dmesg.zip)

Joshua-Riek, 2023-09-06:
Remove the lines i told you above and try adding this to `/etc/modprobe.d/blacklist.conf` and then reboot your system. 
```
blacklist pgdrv
blacklist btusb
blacklist btrtl
blacklist btbcm
```

c0h1b4, 2023-09-07:
Same as before. now NO WiFi and NO Bluetooth.

lspci
0002:20:00.0 PCI bridge: Rockchip Electronics Co., Ltd Device 3588 (rev 01)
0002:21:00.0 Network controller: Intel Corporation Wi-Fi 6 AX210/AX211/AX411 160MHz (rev 1a)
0003:30:00.0 PCI bridge: Rockchip Electronics Co., Ltd Device 3588 (rev 01)
0003:31:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0004:40:00.0 PCI bridge: Rockchip Electronics Co., Ltd Device 3588 (rev 01)
0004:41:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)

uname -a
Linux coh1b4-orange 5.10.160-rockchip #12 SMP Wed Aug 30 00:14:17 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux

[dmesg.zip](https://github.com/Joshua-Riek/ubuntu-rockchip/files/12547144/dmesg.zip)

Joshua-Riek, 2023-09-07:
I tested this out on my board and can not reproduce the issue. WiFi and Bluetooth are working without any modifications on a fresh system install.

```
ubuntu@ubuntu:~$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: enP4p65s0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether c0:74:2b:ff:6f:45 brd ff:ff:ff:ff:ff:ff
3: enP3p49s0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether c0:74:2b:ff:6f:44 brd ff:ff:ff:ff:ff:ff
4: wlP2p33s0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether f4:ce:23:05:a9:10 brd ff:ff:ff:ff:ff:ff
ubuntu@ubuntu:~$ hciconfig -a
hci0:	Type: Primary  Bus: USB
	BD Address: F4:CE:23:05:A9:14  ACL MTU: 1021:4  SCO MTU: 96:6
	UP RUNNING 
	RX bytes:18311 acl:0 sco:0 events:2946 errors:0
	TX bytes:728112 acl:0 sco:0 commands:2944 errors:0
	Features: 0xbf 0xfe 0x0f 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH SNIFF 
	Link mode: PERIPHERAL ACCEPT 
	Name: 'ubuntu'
	Class: 0x000000
	Service Classes: Unspecified
	Device Class: Miscellaneous, 
	HCI Version:  (0xc)  Revision: 0x35fc
	LMP Version:  (0xc)  Subversion: 0x35fc
	Manufacturer: Intel Corp. (2)

ubuntu@ubuntu:~$ lspci
0002:20:00.0 PCI bridge: Rockchip Electronics Co., Ltd Device 3588 (rev 01)
0002:21:00.0 Network controller: Intel Corporation Wi-Fi 6 AX210/AX211/AX411 160MHz (rev 1a)
0003:30:00.0 PCI bridge: Rockchip Electronics Co., Ltd Device 3588 (rev 01)
0003:31:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0004:40:00.0 PCI bridge: Rockchip Electronics Co., Ltd Device 3588 (rev 01)
0004:41:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
ubuntu@ubuntu:~$ uname -a
Linux ubuntu 5.10.160-rockchip #12 SMP Wed Aug 30 00:14:17 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux
ubuntu@ubuntu:~$ 
```

closed 2023-09-09 by Joshua-Riek

duboisp, 2023-12-24:
I had the same issue with my opi 5 plus + AX210 and I did found a solution that worked for me. I did found a few line with errors related to iwlwifi in my dmsg.

To enable the wifi, from a terminal execute:

```shell
sudo mv /usr/lib/firmware/iwlwifi-ty-a0-gf-a0.pnvm  /usr/lib/firmware/iwlwifi-ty-a0-gf-a0.bak
```
then reboot.

Note: That file might require to be move every time after a system update.

Reference: https://askubuntu.com/a/1360291 and https://bugzilla.kernel.org/show_bug.cgi?id=212371#c13