Joshua-Riek/ubuntu-rockchip #355: [Rock5b] Ubuntu headset selection dialog started appearing/disappearing randomly

watou, 2023-09-08:
At latest package level, the Ubuntu headset selection dialog started appearing/disappearing randomly and so quickly it was almost impossible to see what the dialog said.  It happened more often when playing YouTube in Chrome but could appear at other times as well.  

I installed the Gnome extension "Auto select headset" and this seems to have succeeded in suppressing the frequent and seemingly random flashing of this dialog.  I want to say this started occurring after you fixed the analog audio driver, but I could have that wrong.  I do know that this issue had only very recently surfaced.

Joshua-Riek, 2023-09-08:
Can you provide the full dmesg output when this happens? Also, please check that your headset is fully connected.

watou, 2023-09-08:
[dmesg.txt](https://github.com/Joshua-Riek/ubuntu-rockchip/files/12563653/dmesg.txt)
I have no headset or anything attached to analog audio.  I only use HDMI0 out.  I turned off the Gnome extension and see the dialog flash again, but I don't see a correlation in the log between the dialog appearing and new lines added to the dmesg log.

closed 2023-10-28 by Joshua-Riek