Joshua-Riek/ubuntu-rockchip #357: Kernel 6.5.0 pre-release

vtv3000, 2023-09-11:
Hi. I cannot boot on orange pi5plus or Ethernet  not work and I cannot communicate via ssh

Joshua-Riek, 2023-09-11:
I specifically asked for no issues to be open in regards to the 6.5.0 kernel as it is heavily a work in progress. 

I know the system boots. Use a serial console to interact with the device.

closed 2023-09-11 by Joshua-Riek

vtv3000, 2023-09-15:
Hi
I understand about 6.5 - it’s hard work I promise you can make it :)

I have a question - do you have kernel 5.19 ? Or maybe instructions how
compiler this kernel ?

Joshua-Riek, 2023-09-15:
The RK3588 SOC was introduced into mainline Linux version 6.2, so using 5.19 would not be possible.

vtv3000, 2023-09-15:
You say “mainline Linux version 6.2”

Why you make 6.5 ? Maybe 6.2 will be easy ?

Joshua-Riek, 2023-09-15:
The 6.5 kernel introduced many features for the RK3588 SOC, using 6.2 would be complex because fewer hardware features are supported.

vtv3000, 2023-09-15:
Ok I understand . Thanks for you hard work

vtv3000, 2023-09-21:
Hi . I understand about hard work about 6.5 .
Can you compile last kernel v5 ?   May be 5.18 …or write instruction for
compile kernel on orange 5 plus  (not cross compile)

5.10 and 5.18 many differents for me .

Joshua-Riek, 2023-09-21:
The 5.x kernel is not compatible with the RK3588 SOC, so this is not possible. There is a 5.x BSP that uses an **Android** kernel which is not real Linux.

vtv3000, 2023-09-21:
it is very sad. how are you doing with kernel 6 development?