Joshua-Riek/ubuntu-rockchip #361: Brand new Orange Pi 5 Freeze

JayJay3, 2023-09-14:
My brand new Orange Pi 5 freeze often, especially when browser is used. This is related #219 which was closed. So I am reporting gain. This happens with other OPi5 OS versions such as Arch, Debian. USB devices connected was memory stick, keyboard, wireless mouse.

file:///home/orangepi/sys.log

JayJay3, 2023-09-14:
Sep 15 05:49:08 orangepi5 kernel: Booting Linux on physical CPU 0x0000000000 [0x412fd050]
Sep 15 05:49:08 orangepi5 kernel: Linux version 5.10.110-rockchip-rk3588 (root@orangepi5) (aarch64-linux-gnu-gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0, GNU ld (GNU Binutils for Ubuntu) 2.38) #1.1.6 SMP Thu Jun 1 21:23:54 CST 2023
Sep 15 05:49:08 orangepi5 kernel: Machine model: Orange Pi 5
Sep 15 05:49:08 orangepi5 kernel: efi: UEFI not found.
Sep 15 05:49:08 orangepi5 kernel: OF: fdt: Reserved memory: failed to reserve memory for node 'drm-logo@00000000': base 0x0000000000000000, size 0 MiB
Sep 15 05:49:08 orangepi5 kernel: OF: fdt: Reserved memory: failed to reserve memory for node 'drm-cubic-lut@00000000': base 0x0000000000000000, size 0 MiB
Sep 15 05:49:08 orangepi5 kernel: Reserved memory: bypass cma node, using cmdline CMA params instead
Sep 15 05:49:08 orangepi5 kernel: OF: reserved mem: node cma compatible matching fail
Sep 15 05:49:08 orangepi5 kernel: Zone ranges:
Sep 15 05:49:08 orangepi5 kernel:   DMA      [mem 0x0000000000200000-0x00000000efffffff]
Sep 15 05:49:08 orangepi5 kernel:   DMA32    empty
Sep 15 05:49:08 orangepi5 kernel:   Normal   empty
Sep 15 05:49:08 orangepi5 kernel: Movable zone start for each node
Sep 15 05:49:08 orangepi5 kernel: Early memory node ranges
Sep 15 05:49:08 orangepi5 kernel:   node   0: [mem 0x0000000000200000-0x00000000efffffff]
Sep 15 05:49:08 orangepi5 kernel: Initmem setup node 0 [mem 0x0000000000200000-0x00000000efffffff]
Sep 15 05:49:08 orangepi5 kernel: On node 0 totalpages: 982528
Sep 15 05:49:08 orangepi5 kernel:   DMA zone: 15352 pages used for memmap
Sep 15 05:49:08 orangepi5 kernel:   DMA zone: 0 pages reserved
Sep 15 05:49:08 orangepi5 kernel:   DMA zone: 982528 pages, LIFO batch:63
Sep 15 05:49:08 orangepi5 kernel: cma: Reserved 128 MiB at 0x00000000e3800000
Sep 15 05:49:08 orangepi5 kernel: psci: probing for conduit method from DT.
Sep 15 05:49:08 orangepi5 kernel: psci: PSCIv1.1 detected in firmware.
Sep 15 05:49:08 orangepi5 kernel: psci: Using standard PSCI v0.2 function IDs
Sep 15 05:49:08 orangepi5 kernel: psci: MIGRATE_INFO_TYPE not supported.
Sep 15 05:49:08 orangepi5 kernel: psci: SMC Calling Convention v1.2
Sep 15 05:49:08 orangepi5 kernel: percpu: Embedded 32 pages/cpu s93416 r8192 d29464 u131072
Sep 15 05:49:08 orangepi5 kernel: pcpu-alloc: s93416 r8192 d29464 u131072 alloc=32*4096
Sep 15 05:49:08 orangepi5 kernel: pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5 [0] 6 [0] 7 
Sep 15 05:49:08 orangepi5 kernel: Detected VIPT I-cache on CPU0
Sep 15 05:49:08 orangepi5 kernel: CPU features: detected: GIC system register CPU interface
Sep 15 05:49:08 orangepi5 kernel: CPU features: detected: Virtualization Host Extensions
Sep 15 05:49:08 orangepi5 kernel: CPU features: detected: ARM errata 1165522, 1319367, or 1530923
Sep 15 05:49:08 orangepi5 kernel: alternatives: patching kernel code
Sep 15 05:49:08 orangepi5 kernel: Built 1 zonelists, mobility grouping on.  Total pages: 967176
Sep 15 05:49:08 orangepi5 kernel: Kernel command line: root=UUID=5e86de0a-e719-461d-9356-763f38ba918a rootwait rootfstype=ext4 splash plymouth.ignore-serial-consoles console=ttyS2,1500000 console=tty1 consoleblank=0 loglevel=1 ubootpart=ad517e0f-3340-a843-8c05-0c93cbd2b0e7 usb-storage.quirks=0x2537:0x1066:u,0x2537:0x1068:u cma=128M  cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1
Sep 15 05:49:08 orangepi5 kernel: Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
Sep 15 05:49:08 orangepi5 kernel: Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
Sep 15 05:49:08 orangepi5 kernel: mem auto-init: stack:off, heap alloc:off, heap free:off
Sep 15 05:49:08 orangepi5 kernel: Memory: 3667812K/3930112K available (16576K kernel code, 3422K rwdata, 6328K rodata, 6848K init, 632K bss, 131228K reserved, 131072K cma-reserved)
Sep 15 05:49:08 orangepi5 kernel: SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=8, Nodes=1
Sep 15 05:49:08 orangepi5 kernel: ftrace: allocating 63271 entries in 248 pages
Sep 15 05:49:08 orangepi5 kernel: ftrace: allocated 248 pages with 5 groups
Sep 15 05:49:08 orangepi5 kernel: rcu: Hierarchical RCU implementation.
Sep 15 05:49:08 orangepi5 kernel: rcu:         RCU event tracing is enabled.
Sep 15 05:49:08 orangepi5 kernel:         Rude variant of Tasks RCU enabled.
Sep 15 05:49:08 orangepi5 kernel:         Tracing variant of Tasks RCU enabled.
Sep 15 05:49:08 orangepi5 kernel: rcu: RCU calculated value of scheduler-enlistment delay is 30 jiffies.
Sep 15 05:49:08 orangepi5 kernel: NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
Sep 15 05:49:08 orangepi5 kernel: GICv3: GIC: Using split EOI/Deactivate mode
Sep 15 05:49:08 orangepi5 kernel: GICv3: 480 SPIs implemented
Sep 15 05:49:08 orangepi5 kernel: GICv3: 0 Extended SPIs implemented
Sep 15 05:49:08 orangepi5 kernel: GICv3: Distributor has no Range Selector support
Sep 15 05:49:08 orangepi5 kernel: GICv3: 16 PPIs implemented
Sep 15 05:49:08 orangepi5 kernel: GICv3: CPU0: found redistributor 0 region 0:0x00000000fe680000
Sep 15 05:49:08 orangepi5 kernel: ITS [mem 0xfe640000-0xfe65ffff]
Sep 15 05:49:08 orangepi5 kernel: ITS@0x00000000fe640000: allocated 8192 Devices @2a00000 (indirect, esz 8, psz 64K, shr 0)
Sep 15 05:49:08 orangepi5 kernel: ITS@0x00000000fe640000: allocated 32768 Interrupt Collections @2a10000 (flat, esz 2, psz 64K, shr 0)
Sep 15 05:49:08 orangepi5 kernel: ITS: using cache flushing for cmd queue
[... cut ...]
Sep 15 05:49:16 orangepi5 systemd[1]: Starting resolvconf-pull-resolved.service...
Sep 15 05:49:16 orangepi5 NetworkManager[1155]: <info>  [1694720956.5721] device (eth0): Activation: successful, device activated.
Sep 15 05:49:16 orangepi5 NetworkManager[1155]: <info>  [1694720956.5737] manager: NetworkManager state is now CONNECTED_GLOBAL
Sep 15 05:49:16 orangepi5 NetworkManager[1155]: <info>  [1694720956.5767] manager: startup complete
Sep 15 05:49:16 orangepi5 dbus-daemon[1904]: [session uid=1000 pid=1904] Activating via systemd: service name='org.freedesktop.i

Joshua-Riek, 2023-09-14:
This sounds like you don't have a proper power supply, especially since you report it will freeze with other distros. Try to use a 5v 4a power supply.

Also, I don't think you're using this distro based on the kernel name in the logs you posted.

JayJay3, 2023-09-14:
There is one 5v4a power supply model. I am using it! I will try Raspberry power supply which is 5v3a. My testing shows that OPi5 uses up to 11 watts when 8 CPU cores work 100%. So Rpi power supply which is 15watt  is sufficient.

Joshua-Riek, 2023-09-14:
I suggest using a different power supply that is 5v 4a. I have 2 Orange Pi 5 boards running as build servers with no issues.

JayJay3, 2023-09-14:
Problem is that they don't make 5v 4a USB C charger. Max 5v is 3a! So there is no way getting another 5v 4a charger now. The seller I bought no longer sells 5v 4a USB C charger. I think Orange Pi has problem here. They cannot buy 5v 4a charger for OPi5.

Joshua-Riek, 2023-09-15:
I can do nothing here, and I recommend trying another 5v 4a power supply.

closed 2023-09-15 by Joshua-Riek

cantalupo555, 2023-09-15:
> Problem is that they don't make 5v 4a USB C charger. Max 5v is 3a! So there is no way getting another 5v 4a charger now. The seller I bought no longer sells 5v 4a USB C charger. I think Orange Pi has problem here. They cannot buy 5v 4a charger for OPi5.

Buy this:
https://aliexpress.com/item/1005003160082010.html

LabEG, 2023-11-21:
I also have a problem with friezes. But I solved it. I have 2 full HD monitors. One is connected via HDMI, the second via type-C via a hub. I installed the official distribution and everything worked smoothly. I installed the Joshua-Riek distribution and had friezes. Then I remembered that in the official distribution the second monitor did not work correctly. At first I thought it was because of the old HDMI cable. I changed the refresh rates from 75 hertz to 60 hertz. And the type-s monitor began to work correctly. Then I changed the screen frequency in the Joshua-Riek distribution from 75 hertz to 60 hertz and everything began to work smoothly.

Here's a fun fact.

If you set both monitors to 60 hertz, the freezes start again.