Joshua-Riek/ubuntu-rockchip #362: Orange Pi 5 Ubuntu screen blank doesn't work

JayJay3, 2023-09-16:
I set up my Orange pi 5 Ubuntu as follows;

Dim screen: on
Screen blank: 5 minutes
Automatic suspend: off
Power button behavior: Power off

But screen doesn't go blank after 5 minutes. Screen never goes blank. I tested automatic suspend. That works. But not screen blank.

JayJay3, 2023-09-17:
I forgot to mention that it's "Gnome" interface. Also that mouse and screen freeze also "Gnome" version. Xfce version works OK. It's not power adapter problem! I think Gnome is new addition to OPi5. So it has bugs.

Joshua-Riek, 2023-09-17:
I think you are confused and not using this distro as I previously stated. I assume you are using the official distros the Orange Pi company created, I have no relation to them, and I will not fix their distros. 

The GitHub repo below is the official build system from the Orange Pi company, you can try to express your issues there but I can almost guarantee you will not get any help.

https://github.com/orangepi-xunlong/orangepi-build

closed 2023-09-17 by Joshua-Riek

JayJay3, 2023-09-17:
Sorry. I didn't know that. I will content with Xfce version for now. If I have to reinstall Ubuntu, I will try your version.

JayJay3, 2023-09-18:
Does your distro support OpenGL ES version 3?
Although Orange Pi claims it supports OpenGL ES version 3, I cann't find any distro that supports ES version 3.

Joshua-Riek, 2023-09-18:
Yes, it uses a hack called panfork for this. The implementation that Orange Pi uses is very broken.

JayJay3, 2023-09-20:
I got new SD card and tried your version. It has more problems than Orange Pi versions. First, Chrome browser crashed. On next boot, asked to send crash report. After some reboots, Chrome worked. But keep asking password. I never seen browser asking account password. Then Ubuntu internal error occurred. Screen blank accompanies log out. After screen blank, Ubuntu keep asking password. So I gave up!

Joshua-Riek, 2023-09-20:
Your power supply is insufficient, or you still are not using this distro. 

Rambling about an extensive list of issues will do nothing. You need to provide reproducible steps with meaningful information.