Joshua-Riek/ubuntu-rockchip #363: Upgrading Orange Pi 5 Plus to mainline kernel

adminy, 2023-09-16:
This device with supposedly superior specs is getting left behind, is there anything that can be done to bump that version? Even at the expense of having it just boot without any of the devices working properly. As this gap is growing each day with plenty of features coming into the new kernel releases that improve many bugs and add many useful features.

Joshua-Riek, 2023-09-16:
I have experimental releases with the 6.5.0 kernel, what exactly are you looking for?

Joshua-Riek, 2023-09-17:
I have some 6.5.0 builds that finished a few hours ago, there is no HDMI output, so you need to use a serial console.

https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/6210593078

mijk84, 2023-09-17:
If I had to guess, they are likely running into the issue I'm having where the LED is solid  green and there is zero activity from the device. For me this happens with any image that has the  mainline kernel.

Joshua-Riek, 2023-09-17:
Use a serial console, to find out what the device is doing, the LED blinks for me and works with no issue.

adminy, 2023-09-23:
I have a 5G modem and a wifi 6E device that I want to get working. They just need to have latest kernel for driver reasons. Thanks for the build link.

closed 2023-09-23 by adminy