Joshua-Riek/ubuntu-rockchip #364: Is it possible to use mpv without DE?

demosspro, 2023-09-17:
Hello dear Joshua!
First - i wanna say you a BIG thanks for your work. It is pretty good images to start work with that SBC.
Can i ask you - is it possible to obtain MPV with normal worked futures on that board with image without DE?

Joshua-Riek, 2023-09-17:
I have no idea, my only experience with MPV is to playback videos in a DE environment. I don't know much about the inner workings of MPV or video decoding/encoding.

demosspro, 2023-09-17:
Joshua thanks for the answer! I will look for those who may know the answer to this question. But maybe they will be here too and can tell us the answer. Their advice may be to turn off DE and use a few specific options to play the video file by MPV using a command from the terminal.