Joshua-Riek/ubuntu-rockchip #365: NanoPi R5C support?

mightymos, 2023-09-18:
So I have a short length M.2 eMMC module removed from an upgraded Steam Desk...and would like to place it in something as a boot drive. The R5C is relatively inexpensive along with a metal case for silent running ($50 USD).

If it makes any difference I could donate hardware or the like.
I know any additional device support is more work...but what are the chances of supporting the R5C?

Joshua-Riek, 2023-09-18:
Right now, I'm only working on RK3588 devices. I would like to expand to others, but this would require me to re-work my existing build process to add support for other Rockchip SOC devices. This could take me a month or two.

closed 2023-09-18 by Joshua-Riek

mightymos, 2023-09-18:
Thanks for the insight and good luck.

genothomas, 2024-10-18:
RK3568 support for NanoPi R5C would be appreciated 😊