Joshua-Riek/ubuntu-rockchip #366: Wi-Fi not working on boot

mightyguava, 2023-09-19:
Brand new Orange Pi 5B. I burned the [ubuntu-22.04.3-preinstalled-server-arm64-orangepi-5b.img.xz](https://github.com/Joshua-Riek/ubuntu-rockchip/releases/download/v1.27/ubuntu-22.04.3-preinstalled-server-arm64-orangepi-5b.img.xz) to eMMC.

The only changes I've made so far are to configure netplan and run an `apt dist-upgrade`.

`/etc/netplan/50-cloud-init.yaml` is configured with
```
# This file is generated from information provided by the datasource.  Changes
# to it will not persist across an instance reboot.  To disable cloud-init's
# network configuration capabilities, write a file
# /etc/cloud/cloud.cfg.d/99-disable-network-config.cfg with the following:
# network: {config: disabled}
network:
    ethernets:
      eth0:
        dhcp4: true
        optional: true
    version: 2
    renderer: networkd
    wifis:
      wlan0:
        dhcp4: true
        optional: true
        access-points:
          "$SSID":
            password: "$PASSWORD"
```

Not sure why this header is there but cloudinit has not been overwriting it across reboots.

At startup, I get a message that says it's waiting for network ready and waits for about 2 minutes. After that boot completes, and I get a link-local IPv6 address but no IPv4 address. Running `systemctl restart systemd-networkd` will cause wifi to connect almost immediately.

Found these error logs but not sure what to do about it. Timestamp 02:30 is boot time, and then timestamp 02:34:43 is when I manually ran `systemctl restart systemd-networkd`

```
$ journalctl -u netplan-wpa-wlan0.service
-- Boot 79bb3e09a0dc4bdabb3217c1592b0726 --
Sep 19 02:30:08 mirror systemd[1]: Started WPA supplicant for netplan wlan0.
Sep 19 02:30:08 mirror wpa_supplicant[603]: Successfully initialized wpa_supplicant
Sep 19 02:30:08 mirror wpa_supplicant[603]: Could not set interface wlan0 flags (UP): Device or resource busy
Sep 19 02:30:08 mirror wpa_supplicant[603]: nl80211: Could not set interface 'wlan0' UP
Sep 19 02:30:08 mirror wpa_supplicant[603]: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Sep 19 02:30:08 mirror wpa_supplicant[603]: Could not set interface wlan0 flags (UP): Device or resource busy
Sep 19 02:30:08 mirror wpa_supplicant[603]: WEXT: Could not set interface 'wlan0' UP
Sep 19 02:30:08 mirror wpa_supplicant[603]: wlan0: Failed to initialize driver interface
Sep 19 02:30:08 mirror wpa_supplicant[603]: wlan0: CTRL-EVENT-DSCP-POLICY clear_all
Sep 19 02:30:08 mirror systemd[1]: netplan-wpa-wlan0.service: Main process exited, code=exited, status=255/EXCEPTION
Sep 19 02:30:08 mirror systemd[1]: netplan-wpa-wlan0.service: Failed with result 'exit-code'.
Sep 19 02:34:43 mirror systemd[1]: Started WPA supplicant for netplan wlan0.
Sep 19 02:34:43 mirror wpa_supplicant[1031]: Successfully initialized wpa_supplicant
Sep 19 02:34:43 mirror wpa_supplicant[1031]: nl80211: kernel reports: Registration to specific type not supported
Sep 19 02:34:45 mirror wpa_supplicant[1031]: wlan0: Trying to associate with SSID 'yulu'
Sep 19 02:34:46 mirror wpa_supplicant[1031]: wlan0: Associated with a0:36:bc:38:86:dc
Sep 19 02:34:46 mirror wpa_supplicant[1031]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Sep 19 02:34:46 mirror wpa_supplicant[1031]: wlan0: WPA: Key negotiation completed with a0:36:bc:38:86:dc [PTK=CCMP GTK=CCMP]
Sep 19 02:34:46 mirror wpa_supplicant[1031]: wlan0: CTRL-EVENT-CONNECTED - Connection to a0:36:bc:38:86:dc completed [id=0 id_str=]
```

```
$ journalctl -u systemd-networkd
Sep 19 02:30:09 mirror systemd[1]: Starting Network Configuration...
Sep 19 02:30:09 mirror systemd-networkd[616]: lo: Link UP
Sep 19 02:30:09 mirror systemd-networkd[616]: lo: Gained carrier
Sep 19 02:30:09 mirror systemd-networkd[616]: Enumeration completed
Sep 19 02:30:09 mirror systemd[1]: Started Network Configuration.
Sep 19 02:30:10 mirror systemd-networkd[616]: wlan0: Link UP
Sep 19 02:30:10 mirror systemd-networkd[616]: wlan0: Gained carrier
Sep 19 02:30:10 mirror systemd-networkd[616]: eth0: Link UP
Sep 19 02:30:12 mirror systemd-networkd[616]: wlan0: Gained IPv6LL
Sep 19 02:34:43 mirror systemd[1]: Stopping Network Configuration...
Sep 19 02:34:43 mirror systemd-networkd[616]: eth0: DHCPv6 lease lost
Sep 19 02:34:43 mirror systemd-networkd[616]: wlan0: DHCPv6 lease lost
Sep 19 02:34:43 mirror systemd[1]: systemd-networkd.service: Deactivated successfully.
Sep 19 02:34:43 mirror systemd[1]: Stopped Network Configuration.
Sep 19 02:34:43 mirror systemd[1]: Starting Network Configuration...
Sep 19 02:34:43 mirror systemd-networkd[1034]: wlan0: Link UP
Sep 19 02:34:43 mirror systemd-networkd[1034]: eth0: Link UP
Sep 19 02:34:43 mirror systemd-networkd[1034]: lo: Link UP
Sep 19 02:34:43 mirror systemd-networkd[1034]: lo: Gained carrier
Sep 19 02:34:43 mirror systemd-networkd[1034]: wlan0: Gained IPv6LL
Sep 19 02:34:43 mirror systemd-networkd[1034]: Enumeration completed
Sep 19 02:34:43 mirror systemd[1]: Started Network Configuration.
Sep 19 02:34:46 mirror systemd-networkd[1034]: wlan0: Gained carrier
Sep 19 02:34:47 mirror systemd-networkd[1034]: wlan0: DHCPv4 address 192.168.1.132/24 via 192.168.1.1
```

It seems something is competing with wpa_supplicant for wlan0 on boot?

mightyguava, 2023-09-19:
More complete logs from `journalctl -x` below:

<details>
<summary>journalctl logs</summary>
ep 19 02:30:08 mirror systemd[1]: Reached target System Time Set.
░░ Subject: A start job for unit time-set.target has finished successfully
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ A start job for unit time-set.target has finished successfully.
░░ 
░░ The job identifier is 9.
Sep 19 02:30:08 mirror kernel: [dhd] dhd_rx_frame: net device is NOT registered. drop event packet
Sep 19 02:30:08 mirror kernel: [dhd] dhd_conf_set_country : set country CN, revision 0
Sep 19 02:30:08 mirror kernel: [dhd] dhd_rx_frame: net device is NOT registered. drop event packet
Sep 19 02:30:08 mirror kernel: [dhd] dhd_conf_set_country : Country code: CN (CN/0)
Sep 19 02:30:08 mirror systemd-udevd[510]: Using default interface naming scheme 'v249'.
Sep 19 02:30:08 mirror kernel: [dhd] Dongle Host Driver, version 101.10.361.20 (wlan=r892223-20220701-3)
                               drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd compiled on Sep 11 2023 at 20:31:55
Sep 19 02:30:08 mirror kernel: [dhd] Register interface [wlan0]  MAC: b8:2d:28:42:2a:f4
Sep 19 02:30:08 mirror kernel: [dhd] dhd_tcpack_suppress_set: TCP ACK Suppress mode 3 -> mode 0
Sep 19 02:30:08 mirror kernel: [dhd] dhd_tcpack_suppress_set: TCPACK_INFO_MAXNUM=10, TCPDATA_INFO_MAXNUM=10
Sep 19 02:30:08 mirror kernel: [dhd] [wlan0] wl_android_wifi_off : in g_wifi_on=1, on_failure=1
Sep 19 02:30:08 mirror kernel: [dhd] dhd_rtt_deinit: ENTER
Sep 19 02:30:08 mirror kernel: [dhd] dhd_rtt_deinit: EXIT, err = 0
Sep 19 02:30:08 mirror kernel: [dhd] dhd_bus_devreset: == Power OFF ==
Sep 19 02:30:08 mirror kernel: [dhd] dhd_bus_stop: making DHD_BUS_DOWN
Sep 19 02:30:08 mirror kernel: [dhd] dhd_dpc_kill: tasklet disabled
Sep 19 02:30:08 mirror kernel: [dhd] dhd_bus_devreset: making DHD_BUS_DOWN
Sep 19 02:30:08 mirror kernel: [dhd] dhd_bus_devreset:  WLAN OFF Done
Sep 19 02:30:08 mirror kernel: [dhd] wifi_platform_set_power = 0, delay: 0 msec
Sep 19 02:30:08 mirror kernel: [dhd] ======== PULL WL_REG_ON(-1) LOW! ========
Sep 19 02:30:08 mirror kernel: [WLAN_RFKILL]: rockchip_wifi_power: 0
Sep 19 02:30:08 mirror kernel: [WLAN_RFKILL]: rockchip_wifi_power: toggle = false
Sep 19 02:30:08 mirror kernel: wifi power off
Sep 19 02:30:08 mirror systemd[1]: Found device /sys/subsystem/net/devices/wlan0.
░░ Subject: A start job for unit sys-subsystem-net-devices-wlan0.device has finished successfully
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ A start job for unit sys-subsystem-net-devices-wlan0.device has finished successfully.
░░ 
░░ The job identifier is 78.
Sep 19 02:30:08 mirror systemd[1]: Started WPA supplicant for netplan wlan0.
░░ Subject: A start job for unit netplan-wpa-wlan0.service has finished successfully
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ A start job for unit netplan-wpa-wlan0.service has finished successfully.
░░ 
░░ The job identifier is 77.
Sep 19 02:30:08 mirror wpa_supplicant[603]: Successfully initialized wpa_supplicant
Sep 19 02:30:08 mirror wpa_supplicant[603]: Could not set interface wlan0 flags (UP): Device or resource busy
Sep 19 02:30:08 mirror wpa_supplicant[603]: nl80211: Could not set interface 'wlan0' UP
Sep 19 02:30:08 mirror wpa_supplicant[603]: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Sep 19 02:30:08 mirror wpa_supplicant[603]: Could not set interface wlan0 flags (UP): Device or resource busy
Sep 19 02:30:08 mirror wpa_supplicant[603]: WEXT: Could not set interface 'wlan0' UP
Sep 19 02:30:08 mirror wpa_supplicant[603]: wlan0: Failed to initialize driver interface
Sep 19 02:30:08 mirror kernel: [dhd] dhd_pri_open : probe is already running! return.
Sep 19 02:30:08 mirror kernel: [dhd] dhd_pri_open : probe is already running! return.
Sep 19 02:30:08 mirror wpa_supplicant[603]: wlan0: CTRL-EVENT-DSCP-POLICY clear_all
Sep 19 02:30:08 mirror kernel: [WLAN_RFKILL]: wifi shut off power [GPIO24-0]
Sep 19 02:30:08 mirror kernel: [dhd] [wlan0] wl_android_wifi_off : out
Sep 19 02:30:08 mirror kernel: [dhd] dhdpcie_pci_probe : mutex is released.
Sep 19 02:30:08 mirror kernel: [dhd] _dhd_module_init: Exit err=0
Sep 19 02:30:08 mirror systemd[1]: netplan-wpa-wlan0.service: Main process exited, code=exited, status=255/EXCEPTION
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
[... 1905 lines cut ...]
░░ Subject: A start job for unit sys-subsystem-net-devices-wlan0.device has finished successfully
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ A start job for unit sys-subsystem-net-devices-wlan0.device has finished successfully.
░░ 
░░ The job identifier is 78.
Sep 19 02:30:08 mirror systemd[1]: Started WPA supplicant for netplan wlan0.
░░ Subject: A start job for unit netplan-wpa-wlan0.service has finished successfully
░░ Defined-By: systemd
░░ Support: http://www.ubuntu.com/support
░░ 
░░ A start job for unit netplan-wpa-wlan0.service has finished successfully.
░░ 
░░ The job identifier is 77.
Sep 19 02:30:08 mirror wpa_supplicant[603]: Successfully initialized wpa_supplicant
Sep 19 02:30:08 mirror wpa_supplicant[603]: Could not set interface wlan0 flags (UP): Device or resource busy
Sep 19 02:30:08 mirror wpa_supplicant[603]: nl80211: Could not set interface 'wlan0' UP
Sep 19 02:30:08 mirror wpa_supplicant[603]: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Sep 19 02:30:08 mirror wpa_supplicant[603]: Could not set interface wlan0 flags (UP): Device or resource busy
Sep 19 02:30:08 mirror wpa_supplicant[603]: WEXT: Could not set interface 'wlan0' UP
Sep 19 02:30:08 mirror wpa_supplicant[603]: wlan0: Failed to initialize driver interface
```

Looks like something is powering off wifi right before wpa_supplicant tries to run?

Joshua-Riek, 2023-09-19:
I wifi should be configured with the network config file in the boot partition before the first boot.

Also, if you install the desktop version, do you have a wifi issue?

mightyguava, 2023-09-19:
As in I should put my netplan YAML changes in `/boot/firmware/network-config` instead? I didn’t see anything in the wiki so I just found some instructions on Google. The use of cloud-init confused me too. What is that being used for?

Will try desktop tomorrow but I’d really like to get server working as I want to give the display device directly to an application instead of the desktop.

mightyguava, 2023-09-19:
> I wifi should be configured with the network config file in the boot partition before the first boot.

Ah I understand now, you are explaining what that header is about. My config is correct looking at https://github.com/Joshua-Riek/ubuntu-rockchip/issues/283.

So after a couple restarts I actually did manage to get wifi working at boot, same `netplan` config file. For conciseness, here's `journalctl -xe | grep wlan0` for the successful run

```
Sep 19 09:07:29 mirror kernel: [dhd] Register interface [wlan0]  MAC: b8:2d:28:42:2a:f4
Sep 19 09:07:29 mirror kernel: [dhd] [wlan0] wl_android_wifi_off : in g_wifi_on=1, on_failure=1
Sep 19 09:07:29 mirror systemd[1]: Found device /sys/subsystem/net/devices/wlan0.
░░ Subject: A start job for unit sys-subsystem-net-devices-wlan0.device has finished successfully
░░ A start job for unit sys-subsystem-net-devices-wlan0.device has finished successfully.
Sep 19 09:07:29 mirror systemd[1]: Started WPA supplicant for netplan wlan0.
░░ Subject: A start job for unit netplan-wpa-wlan0.service has finished successfully
░░ A start job for unit netplan-wpa-wlan0.service has finished successfully.
Sep 19 09:07:29 mirror kernel: [dhd] [wlan0] wl_android_wifi_off : out
Sep 19 09:07:29 mirror kernel: [dhd] [wlan0] dhd_open : Enter
Sep 19 09:07:29 mirror kernel: [dhd] [wlan0] wl_android_wifi_on : in g_wifi_on=0
Sep 19 09:07:30 mirror kernel: [dhd] [wlan0] wl_android_wifi_on : Success
Sep 19 09:07:30 mirror kernel: [dhd] [wlan0] dhd_open : Exit ret=0
Sep 19 09:07:30 mirror kernel: [dhd] [wlan0] dhd_pri_open : tx queue started
Sep 19 09:07:30 mirror kernel: [dhd] [wlan0] custom_xps_map_set : Done. mapping cpu
Sep 19 09:07:30 mirror systemd-networkd[620]: wlan0: Link UP
Sep 19 09:07:30 mirror systemd-networkd[620]: wlan0: Link DOWN
Sep 19 09:07:30 mirror systemd-networkd[620]: wlan0: Re-configuring with /run/systemd/network/10-netplan-wlan0.network
Sep 19 09:07:30 mirror systemd-networkd[620]: wlan0: Link UP
Sep 19 09:07:30 mirror systemd-networkd[620]: wlan0: Gained carrier
Sep 19 09:07:30 mirror systemd-networkd[620]: wlan0: Lost carrier
Sep 19 09:07:30 mirror systemd-networkd[620]: wlan0: DHCPv6 lease lost
Sep 19 09:07:32 mirror systemd-networkd[620]: wlan0: Gained IPv6LL
Sep 19 09:07:33 mirror wpa_supplicant[606]: wlan0: Trying to associate with SSID 'my-ssid'
Sep 19 09:07:33 mirror kernel: [dhd] [wlan0] wl_ext_set_chanspec : channel 5g-161(0xe39b)
Sep 19 09:07:33 mirror kernel: [dhd] [wlan0] wl_conn_debug_info : Connecting with a0:36:bc:38:86:dc ssid "my-ssid", len (4), channel=5g-161(chan_cnt=1), sec=wpa2/psk/mfpn/aes, rssi=-31
Sep 19 09:07:33 mirror kernel: [dhd] [wlan0] wl_iw_event : Link UP with a0:36:bc:38:86:dc
Sep 19 09:07:33 mirror kernel: [dhd] [wlan0] wl_ext_iapsta_event : [S] Link UP with a0:36:bc:38:86:dc
Sep 19 09:07:33 mirror kernel: [dhd] [wlan0] wl_bss_connect_done : Report connect result - connection succeeded
Sep 19 09:07:33 mirror wpa_supplicant[606]: wlan0: Associated with a0:36:bc:38:86:dc
Sep 19 09:07:33 mirror wpa_supplicant[606]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Sep 19 09:07:33 mirror kernel: [dhd] [wlan0] wl_add_keyext : key index (0)
Sep 19 09:07:33 mirror wpa_supplicant[606]: wlan0: WPA: Key negotiation completed with a0:36:bc:38:86:dc [PTK=CCMP GTK=CCMP]
Sep 19 09:07:33 mirror wpa_supplicant[606]: wlan0: CTRL-EVENT-CONNECTED - Connection to a0:36:bc:38:86:dc completed [id=0 id_str=]
Sep 19 09:07:33 mirror systemd-networkd[620]: wlan0: Gained carrier
Sep 19 09:07:34 mirror systemd-networkd[620]: wlan0: DHCPv4 address 192.168.1.132/24 via 192.168.1.1
Sep 19 09:09:30 mirror cloud-init[675]: ci-info: | wlan0  |  True |        192.168.1.132         | 255.255.255.0 | global | b8:2d:28:42:2a:f4 |
Sep 19 09:09:30 mirror cloud-init[675]: ci-info: | wlan0  |  True | fe80::ba2d:28ff:fe42:2af4/64 |       .       |  link  | b8:2d:28:42:2a:f4 |
Sep 19 09:09:30 mirror cloud-init[675]: ci-info: |   0   |   0.0.0.0    | 192.168.1.1 |     0.0.0.0     |   wlan0   |   UG  |
Sep 19 09:09:30 mirror cloud-init[675]: ci-info: |   1   | 192.168.1.0  |   0.0.0.0   |  255.255.255.0  |   wlan0   |   U   |
Sep 19 09:09:30 mirror cloud-init[675]: ci-info: |   2   | 192.168.1.1  |   0.0.0.0   | 255.255.255.255 |   wlan0   |   UH  |
Sep 19 09:09:30 mirror cloud-init[675]: ci-info: |   3   | 192.168.1.5  |   0.0.0.0   | 255.255.255.255 |   wlan0   |   UH  |
Sep 19 09:09:30 mirror cloud-init[675]: ci-info: |   4   | 192.168.1.10 |   0.0.0.0   | 255.255.255.255 |   wlan0   |   UH  |
Sep 19 09:09:30 mirror cloud-init[675]: ci-info: |   1   |  fe80::/64  |    ::   |   wlan0   |   U   |
Sep 19 09:09:30 mirror cloud-init[675]: ci-info: |   3   |    local    |    ::   |   wlan0   |   U   |
Sep 19 09:09:30 mirror cloud-init[675]: ci-info: |   4   |  multicast  |    ::   |   wlan0   |   U   |
```

And a subsequent restart that failed to connect to wifi, with no config changes.

```
Sep 19 09:11:21 mirror kernel: [dhd] Register interface [wlan0]  MAC: b8:2d:28:42:2a:f4
Sep 19 09:11:21 mirror kernel: [dhd] [wlan0] wl_android_wifi_off : in g_wifi_on=1, on_failure=1
Sep 19 09:11:21 mirror systemd[1]: Found device /sys/subsystem/net/devices/wlan0.
░░ Subject: A start job for unit sys-subsystem-net-devices-wlan0.device has finished successfully
░░ A start job for unit sys-subsystem-net-devices-wlan0.device has finished successfully.
Sep 19 09:11:21 mirror systemd[1]: Started WPA supplicant for netplan wlan0.
[... cut ...]
Sep 19 09:13:22 mirror cloud-init[658]: ci-info: | wlan0  |  True | fe80::ba2d:28ff:fe42:2af4/64 |     .     |  link | b8:2d:28:42:2a:f4 |
Sep 19 09:13:22 mirror cloud-init[658]: ci-info: |   1   |  fe80::/64  |    ::   |   wlan0   |   U   |
Sep 19 09:13:22 mirror cloud-init[658]: ci-info: |   3   |    local    |    ::   |   wlan0   |   U   |
Sep 19 09:13:22 mirror cloud-init[658]: ci-info: |   4   |  multicast  |    ::   |   wlan0   |   U   |
```

It sure seems like there's a race with something occupying `wlan0` the same time that `netplan-wpa-wlan0.service` tries to initialize it.

mightyguava, 2023-09-19:
The suspect being this block

```
Sep 19 09:07:29 mirror kernel: [dhd] [wlan0] wl_android_wifi_off : out
Sep 19 09:07:29 mirror kernel: [dhd] [wlan0] dhd_open : Enter
Sep 19 09:07:29 mirror kernel: [dhd] [wlan0] wl_android_wifi_on : in g_wifi_on=0
Sep 19 09:07:30 mirror kernel: [dhd] [wlan0] wl_android_wifi_on : Success
Sep 19 09:07:30 mirror kernel: [dhd] [wlan0] dhd_open : Exit ret=0
Sep 19 09:07:30 mirror kernel: [dhd] [wlan0] dhd_pri_open : tx queue started
Sep 19 09:07:30 mirror kernel: [dhd] [wlan0] custom_xps_map_set : Done. mapping cpu
```

If it appears before `wpa_supplicant`, wifi works on boot. If it appears after, wifi doesn't work on boot as `wpa_supplicant` can't access `wlan0`.

mightyguava, 2023-09-19:
Following instructions from ChatGPT 🤖 , I ran 

```
systemctl edit netplan-wpa-wlan0.service
```

and added

```
[Service]
ExecStartPre=/bin/sleep 1
```

to add a 1 second delay before `wpa_supplicant`. Now my wifi starts consistently across reboots.

The remaining issue is that `systemd-networkd-wait-online` still thinks network isn't available and waits the full 2 minutes, even though you can clearly see that wlan0 is getting a DHCP address, and NTP is working.

Logs below

<details>
<summary>journalctl -b (collapsed)</summary>

```
Sep 19 10:07:50 mirror kernel: [dhd] Register interface [wlan0]  MAC: b8:2d:28:42:2a:f4
Sep 19 10:07:50 mirror kernel: [dhd] dhd_tcpack_suppress_set: TCP ACK Suppress mode 3 -> mode 0
Sep 19 10:07:50 mirror kernel: [dhd] dhd_tcpack_suppress_set: TCPACK_INFO_MAXNUM=10, TCPDATA_INFO_MAXNUM=10
Sep 19 10:07:50 mirror kernel: [dhd] [wlan0] wl_android_wifi_off : in g_wifi_on=1, on_failure=1
Sep 19 10:07:50 mirror kernel: [dhd] dhd_rtt_deinit: ENTER
Sep 19 10:07:50 mirror kernel: [dhd] dhd_rtt_deinit: EXIT, err = 0
Sep 19 10:07:50 mirror kernel: [dhd] dhd_bus_devreset: == Power OFF ==
Sep 19 10:07:50 mirror kernel: [dhd] dhd_bus_stop: making DHD_BUS_DOWN
Sep 19 10:07:50 mirror kernel: [dhd] dhd_dpc_kill: tasklet disabled
Sep 19 10:07:50 mirror kernel: [dhd] dhd_bus_devreset: making DHD_BUS_DOWN
Sep 19 10:07:50 mirror kernel: [dhd] dhd_bus_devreset:  WLAN OFF Done
Sep 19 10:07:50 mirror kernel: [dhd] wifi_platform_set_power = 0, delay: 0 msec
Sep 19 10:07:50 mirror kernel: [dhd] ======== PULL WL_REG_ON(-1) LOW! ========
Sep 19 10:07:50 mirror kernel: [WLAN_RFKILL]: rockchip_wifi_power: 0
Sep 19 10:07:50 mirror kernel: [WLAN_RFKILL]: rockchip_wifi_power: toggle = false
Sep 19 10:07:50 mirror kernel: wifi power off
Sep 19 10:07:50 mirror systemd[1]: Found device /sys/subsystem/net/devices/wlan0.
Sep 19 10:07:50 mirror systemd[1]: Starting WPA supplicant for netplan wlan0...
Sep 19 10:07:50 mirror kernel: [WLAN_RFKILL]: wifi shut off power [GPIO24-0]
Sep 19 10:07:50 mirror kernel: [dhd] [wlan0] wl_android_wifi_off : out
Sep 19 10:07:50 mirror kernel: [dhd] dhdpcie_pci_probe : mutex is released.
Sep 19 10:07:50 mirror kernel: [dhd] _dhd_module_init: Exit err=0
Sep 19 10:07:51 mirror cloud-init[607]: Cloud-init v. 23.2.2-0ubuntu0~22.04.1 running 'init-local' at Tue, 19 Sep 2023 10:07:50 +0000. Up 8.87 seconds.
Sep 19 10:07:51 mirror systemd[1]: Finished Initial cloud-init job (pre-networking).
Sep 19 10:07:51 mirror systemd[1]: Reached target Preparation for Network.
Sep 19 10:07:51 mirror systemd[1]: Starting Raise network interfaces...
Sep 19 10:07:51 mirror systemd[1]: Starting Network Configuration...
Sep 19 10:07:51 mirror ifup[640]: /etc/network/if-up.d/resolved: 12: mystatedir: not found
Sep 19 10:07:51 mirror systemd-networkd[616]: lo: Link UP
Sep 19 10:07:51 mirror systemd-networkd[616]: lo: Gained carrier
Sep 19 10:07:51 mirror systemd-networkd[616]: Enumeration completed
Sep 19 10:07:51 mirror systemd[1]: Started Network Configuration.
Sep 19 10:07:51 mirror kernel: [dhd] dhd_query_bus_erros: Dongle Reset occurred, cannot proceed
Sep 19 10:07:51 mirror kernel: [dhd] [Repeats 0 times]
Sep 19 10:07:51 mirror kernel: [dhd] CFG80211-ERROR) wl_cfg80211_dump_station : WLC_GET_ASSOCLIST error -19
Sep 19 10:07:51 mirror kernel: [dhd] dhd_pri_open : no mutex held
Sep 19 10:07:51 mirror kernel: [dhd] dhd_pri_open : set mutex lock
Sep 19 10:07:51 mirror kernel: [dhd] [wlan0] dhd_open : Enter
Sep 19 10:07:51 mirror kernel: [dhd] Dongle Host Driver, version 101.10.361.20 (wlan=r892223-20220701-3)
                               drivers/net/wireless/rockchip_wlan/rkwifi/bcmdhd compiled on Sep 11 2023 at 20:31:55
Sep 19 10:07:51 mirror kernel: [dhd] dhd_open: ######### called for ifidx=0 #########
Sep 19 10:07:51 mirror kernel: [dhd] [wlan0] wl_android_wifi_on : in g_wifi_on=0
Sep 19 10:07:51 mirror kernel: [dhd] wifi_platform_set_power = 1, delay: 200 msec
Sep 19 10:07:51 mirror kernel: [dhd] ======== PULL WL_REG_ON(-1) HIGH! ========
Sep 19 10:07:51 mirror kernel: [WLAN_RFKILL]: rockchip_wifi_power: 1
Sep 19 10:07:51 mirror kernel: [WLAN_RFKILL]: rockchip_wifi_power: toggle = false
Sep 19 10:07:51 mirror systemd[1]: Starting Wait for Network to be Configured...
[... 167 lines cut ...]
Sep 19 10:09:50 mirror cloud-init[672]: ci-info: | wlan0  |  True |        192.168.1.132         | 255.255.255.0 | global | b8:2d:28:42:2a:f4 |
Sep 19 10:09:50 mirror cloud-init[672]: ci-info: | wlan0  |  True | fe80::ba2d:28ff:fe42:2af4/64 |       .       |  link  | b8:2d:28:42:2a:f4 |
Sep 19 10:09:50 mirror cloud-init[672]: ci-info: +--------+-------+------------------------------+---------------+--------+-------------------+
Sep 19 10:09:50 mirror cloud-init[672]: ci-info: ++++++++++++++++++++++++++++++Route IPv4 info+++++++++++++++++++++++++++++++
Sep 19 10:09:50 mirror cloud-init[672]: ci-info: +-------+--------------+-------------+-----------------+-----------+-------+
Sep 19 10:09:50 mirror cloud-init[672]: ci-info: | Route | Destination  |   Gateway   |     Genmask     | Interface | Flags |
Sep 19 10:09:50 mirror cloud-init[672]: ci-info: +-------+--------------+-------------+-----------------+-----------+-------+
Sep 19 10:09:50 mirror cloud-init[672]: ci-info: |   0   |   0.0.0.0    | 192.168.1.1 |     0.0.0.0     |   wlan0   |   UG  |
Sep 19 10:09:50 mirror cloud-init[672]: ci-info: |   1   | 192.168.1.0  |   0.0.0.0   |  255.255.255.0  |   wlan0   |   U   |
[... cut ...]
Sep 19 10:09:50 mirror cloud-init[672]: 2023-09-19 10:09:50,712 - schema.py[DEPRECATED]: Deprecated cloud-config provided:
Sep 19 10:09:50 mirror cloud-init[672]: chpasswd.list: List of ``username:password`` pairs. Each user will have the corresponding password set. A password can be randomly generated by specifying ``RANDOM`` or ``R`` as a user's password. A hashed password, created by a tool like ``mkpasswd``, can be specified. A regex (``r'\$(1|2a|2y|5|6)(\$.+){2}'``) is used to determine if a password value should be treated as a hash. Deprecated in version 22.2. Use ``users`` instead.
```
</details>

mightyguava, 2023-09-19:
I ended up changing the timeout for `systemd-networkd-wait-online` to 10 seconds instead of the default 2 minutes. There's nothing in the logs as to _why_ it's failing. Any idea where to file a bug against?

Neither the 1 second delay in `netplan-wpa-wlan0.service` nor lowering the timeout for the network wait feels right nor satisfying...

Joshua-Riek, 2023-09-19:
Thanks for putting some time into this, I don't have a lot of time right now, I'd like to make sure wifi is working normally.

Now I do know that the 2 minute wait is a normal Ubuntu 22.04 thing (however, i find it very annoying, if its safe to set it to 10 seconds i will probably set that as the default configuration). It used to be 5 minutes in Ubuntu 20.04

I did try to edit the delay to 10 seconds over the weekend but did not spend much time on it. Did you just set a timeout in the systemd service?

mightyguava, 2023-09-19:
Yeah I did

```
sudo systemctl edit --full systemd-networkd-wait-online.service
```

And edited the command to

```
ExecStart=/usr/lib/systemd/systemd-networkd-wait-online --timeout=10
```

been stable across a few restarts.

Theoretically this service is supposed to exit successfully as soon as one of the interfaces connects to internet. Does it always hang and fail after 2 minutes for you as well?

Joshua-Riek, 2023-09-19:
I did some short testing, and the 2-minute time-out does not happen on my Orange Pi 5B. However, I encountered this on a different device over the weekend, so I know this exists. I don't have time to dig deeper right now, so I can't provide much information.

Joshua-Riek, 2023-09-20:
I made a change to overwrite the default link-local address https://github.com/Joshua-Riek/ubuntu-rockchip/commit/6eda35739053c6ac02bdf39e754e5fbc7a0c3a91. Maybe this will help.

build: https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/6254729259

reference: https://askubuntu.com/questions/1421785/a-start-job-is-running-for-wait-for-network-to-be-configured-because-ipv6-blocke

mightyguava, 2023-09-21:
Oohhhh that makes sense. Great catch. I do not have ipv6.

I switched over to the Ubuntu desktop version and WiFi has been working fine. Getting some more sd cards in a couple days. I’ll flash and test your fix then.

mightyguava, 2023-09-22:
I flashed your new image and double checked that link local setting. Still getting a 2 minute wait on start.

Joshua-Riek, 2023-09-22:
Whats the output of the command `sudo networkctl`?

Joshua-Riek, 2023-09-22:
Also, have you connected to ethernet?

mightyguava, 2023-09-22:
```
ubuntu@ubuntu:~$ sudo networkctl
IDX LINK TYPE     OPERATIONAL SETUP
  1 lo   loopback carrier     unmanaged
  2 eth0 ether    routable    configured
```

I was on wifi only earlier. Now I'm on ethernet. Weirdly when I connect to ethernet, the wifi device fails to initialize.

mightyguava, 2023-09-22:
Okay, now my wlan0 device is refusing to get created at all even without internet. Haven't made any config changes since the last update. It works just fine with the desktop image still so don't think it's a hardware issue.

mightyguava, 2023-09-22:
This is reaaally weird, I don't see `dhd` in my logs anywhere. `dmesg | grep dhd` is empty. Not in `journalctl` either. Is that like my install got corrupted?

Joshua-Riek, 2023-09-22:
> Okay, now my wlan0 device is refusing to get created at all even without internet. Haven't made any config changes since the last update. It works just fine with the desktop image still so don't think it's a hardware issue.

I think you downloaded the Orange Pi 5 image and not the 5B image. 

I have been trying to reproduce this on my side with no luck yet.

mightyguava, 2023-09-23:
I actually just went and re-flashed the SD card half an hour ago and still have not been getting it to work. Ran this

```
sudo dd bs=4194304 if=ubuntu-22.04.3-preinstalled-server-arm64-orangepi-5b.img of=/dev/disk4
```

Put the sd card in. No `wlan0` showing up in `/sys/class/net`.

Gotta go grab dinner. That's probably all the time I have for the day, will look more tomorrow.

mightyguava, 2023-09-23:
The weirdest part is that wifi worked on the first boot with your new image. But it hasn’t worked afterwards even after re-imaging

mightyguava, 2023-09-23:
Managed to get my `wlan0` interface to show up again: the `dhd` stuff doesn't show up if I have my Anker USB hub (which has a keyboard & mouse attached) into the lower (closer to the board) of 2 USB-3 sockets. It's fine connected to the upper one, and connecting the keyboard by itself is fine in either socket. I suppose it's probably some obscure kernel bug I'm hitting.

mightyguava, 2023-09-23:
Diffing the dmesg between the good & bad runs, I see these in the bad run

```
+rk-pcie fe190000.pcie: PCIe Link Fail
+rk-pcie fe190000.pcie: failed to initialize host
```

which seems to lead to the rest of the failures.

<details>
<summary>dmesg diff</summary>

```diff
--- dmesg-cut.good.txt	Fri Sep 22 22:18:31 2023
+++ dmesg-cut.bad.txt	Fri Sep 22 22:18:18 2023
@@ -165,24 +165,24 @@
 HugeTLB registered 32.0 MiB page size, pre-allocated 0 pages
 HugeTLB registered 2.00 MiB page size, pre-allocated 0 pages
 HugeTLB registered 64.0 KiB page size, pre-allocated 0 pages
-raid6: neonx8   gen()  5667 MB/s
-raid6: neonx8   xor()  4411 MB/s
-raid6: neonx4   gen()  5823 MB/s
-raid6: neonx4   xor()  4425 MB/s
-raid6: neonx2   gen()  5341 MB/s
-raid6: neonx2   xor()  4104 MB/s
-raid6: neonx1   gen()  4465 MB/s
-raid6: neonx1   xor()  3441 MB/s
-raid6: int64x8  gen()  1389 MB/s
-raid6: int64x8  xor()   905 MB/s
-raid6: int64x4  gen()  1712 MB/s
-raid6: int64x4  xor()   932 MB/s
-raid6: int64x2  gen()  2506 MB/s
-raid6: int64x2  xor()  1376 MB/s
-raid6: int64x1  gen()  2086 MB/s
-raid6: int64x1  xor()  1017 MB/s
-raid6: using algorithm neonx4 gen() 5823 MB/s
-raid6: .... xor() 4425 MB/s, rmw enabled
+raid6: neonx8   gen()  5657 MB/s
+raid6: neonx8   xor()  4392 MB/s
+raid6: neonx4   gen()  5799 MB/s
+raid6: neonx4   xor()  4405 MB/s
+raid6: neonx2   gen()  5318 MB/s
+raid6: neonx2   xor()  4086 MB/s
+raid6: neonx1   gen()  4445 MB/s
+raid6: neonx1   xor()  3425 MB/s
+raid6: int64x8  gen()  1374 MB/s
+raid6: int64x8  xor()   913 MB/s
+raid6: int64x4  gen()  1704 MB/s
+raid6: int64x4  xor()   928 MB/s
+raid6: int64x2  gen()  2497 MB/s
+raid6: int64x2  xor()  1369 MB/s
+raid6: int64x1  gen()  2075 MB/s
+raid6: int64x1  xor()  1013 MB/s
+raid6: using algorithm neonx4 gen() 5799 MB/s
+raid6: .... xor() 4405 MB/s, rmw enabled
 raid6: using neon recovery algorithm
 fiq_debugger fiq_debugger.0: IRQ fiq not found
 fiq_debugger fiq_debugger.0: IRQ wakeup not found
@@ -305,10 +305,10 @@
 ntfs: driver 2.1.32 [Flags: R/W
 NET: Registered protocol family 38
 xor: measuring software checksum speed
-   8regs           :  6494 MB/sec
-   32regs          :  8615 MB/sec
-   arm64_neon      : 10509 MB/sec
-xor: using function: arm64_neon (10509 MB/sec)
+   8regs           :  6482 MB/sec
+   32regs          :  8608 MB/sec
+   arm64_neon      : 10598 MB/sec
+xor: using function: arm64_neon (10598 MB/sec)
[... 616 lines cut ...]
-[dhd
 rk_gmac-dwmac fe1c0000.ethernet eth0: PHY [stmmac-1:01
 dwmac4: Master AXI performs any burst length
 rk_gmac-dwmac fe1c0000.ethernet eth0: No Safety Features support found
@@ -1399,9 +1239,6 @@
 usb usb8: SerialNumber: xhci-hcd.9.auto
 hub 8-0:1.0: USB hub found
 hub 8-0:1.0: 1 port detected
-[dhd
-[dhd
-[dhd
 squashfs: version 4.0 (2009/01/31) Phillip Lougher
 [BT_RFKILL
 ttyFIQ ttyFIQ0: tty_port_close_start: tty->count = 1 port count = 2
@@ -1419,7 +1256,5 @@
 Bluetooth: BNEP (Ethernet Emulation) ver 1.3
 Bluetooth: BNEP filters: protocol multicast
 Bluetooth: BNEP socket layer initialized
-[dhd
-[dhd
-[dhd
+systemd-journald[453
 ```

I thought about this maybe being a power draw issue, but the hub is only powering a keyboard & mouse... and I don't have any of these problems when booting into the desktop image... I assume the destkop image has the same drivers and kernel as the server image...?

Joshua-Riek, 2023-09-23:
I would take all and any kernel messages with a grain of salt. The kernel is an absolute hackjob of a mess, but since I can not reproduce this, I would say your power supply is bad. Or your router has some weird internet configuration.

mightyguava, 2023-09-23:
The power supply being suspect makes sense, but I'm having the same issue with both a brand new OrangePi 5V 4A charger as well as an Anker 65W PD charger. Router issues shouldn't cause the `wlan0` interface to not initialize.

My router is OPNSense running a standard configuration, no VLANs or other fancy stuff, just DHCPv4.

Anyways, here's the `networkctl` output you asked for running your new image.

```
ubuntu@ubuntu:~$ sudo networkctl
IDX LINK  TYPE     OPERATIONAL SETUP
  1 lo    loopback carrier     unmanaged
  2 eth0  ether    routable    configured
  3 wlan0 wlan     off         unmanaged
```

Still had to wait for that 2 minutes...

Joshua-Riek, 2023-09-23:
I'm not sure what I can do since I can not reproduce this issue on my end. 

Now, I could add the below as default, but I fear this could cause issues with other network configurations.
```
ExecStart=/usr/lib/systemd/systemd-networkd-wait-online --timeout=10
```

mightyguava, 2023-09-23:
Outside of the pcie weirdness, I think there were 2 concrete things:

Add an override for `netplan-wpa-wlan0.service`

```
[Service]
ExecStartPre=/bin/sleep 1
```

And the timeout=10 you had above yeah. Thanks for helping me debug these issues!

Joshua-Riek, 2023-09-23:
Of course, but to confirm, is wifi working for you without any modification?

mightyguava, 2023-09-23:
No.

### After flashing the image, configuring netplan with my SSID/pw, no other modifications
WiFi does not work on boot or restarts. I have to manually run `systemctl restart systemd-netword` after the system boots.

### Adding the `/bin/sleep 1` override to `netplan-wpa-wlan0.service`
WiFi is working after boot, but blocks 2 minutes on `systemd-networkd-wait-online`

### Further adding `--timeout=10` to `systemd-networkd-wait-online`
WiFi works after boot, but blocks 10 seconds on `systemd-networkd-wait-online`.

Joshua-Riek, 2023-09-27:
I did some testing on my Orange Pi 5B and could not reproduce this issue. WiFi works as expected on the server image. My guess is your network-config is not setup correctly.

mightyguava, 2023-09-28:
My network-config was posted at the start of this issue and I have not changed it. Since it’s a timing issue I wonder if my install being on emmc might contribute to it. I’m happy to move on though as there are workarounds and things are working now. Thanks for your help!

Joshua-Riek, 2023-10-02:
Hey, I'm sorry you're still having an issue with this, but I can not reproduce it as it works fine on my end. So I will be closing this.

closed 2023-10-02 by Joshua-Riek

phisch, 2024-04-30:
I have the same issue on my Orange Pi 5B. Running `netplan apply` after boot connects just fine to my wifi, but it does not connect at boot.