Joshua-Riek/ubuntu-rockchip #368: Firmware atheros

Ukhupacha, 2023-09-19:
Hello Joshua,
I was wondering how hard is it to incorporate the firmware-atheros [https://pkgs.org/download/firmware-atheros](url) Many of the rockchip boards do not have an incorporated wifi card, so it is necessary to use usb dongles for wifi.

Thank you for all the images and amazing work.

Joshua-Riek, 2023-09-27:
I would prefer to avoid porting a debian package over to ubuntu. I maintain multiple already, so I dont think this is something I will do.

closed 2023-09-27 by Joshua-Riek

Ferenc-, 2023-09-28:
@Ukhupacha The way I see, people just go ahead and install the Bullseye package on Jammy and it might just work for them? https://gist.github.com/indeedhat/68c62e721f00ea0f29b87bcc81d747d4