Joshua-Riek/ubuntu-rockchip #371: 8k HDMI2.1 output

zjeng, 2023-09-21:
Hi, has anyone seen 8k resolution working on the HDMI output of a ROCK 5B? Our monitor is capable of up to 7680x4320p60Hz. We run Ubuntu in X11 mode and have tested w/ and w/o rock-5b-hdmi1-8k.dts overlay enabled. 
Unfortunately we fall back to 4kp60 all the time. In the log we can see that the 8k attemp comes back with an error but have no idea what the cause is:

`Sep 21 12:27:42 rock-5b /usr/libexec/gdm-x-session[2005]: (II) modeset(0): Allocate new frame buffer 7680x4320 stride
Sep 21 12:27:42 rock-5b kernel: rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_disable] Crtc atomic disable vp0
Sep 21 12:27:42 rock-5b kernel: rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_enable] Update mode to 7680x4320p30, type: 11(if:800, flag:0x0) for vp0 dclk: 1188000000
Sep 21 12:27:42 rock-5b kernel: rockchip-hdptx-phy-hdmi fed60000.hdmiphy: hdptx_ropll_cmn_config bus_width:42625a00 rate:742500
Sep 21 12:27:42 rock-5b kernel: rockchip-hdptx-phy-hdmi fed60000.hdmiphy: hdptx phy pll locked!
Sep 21 12:27:42 rock-5b kernel: rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_enable] dclk_out0 div: 0 dclk_core0 div: 0
Sep 21 12:27:42 rock-5b kernel: rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_enable] set dclk_vop0 to 297000000, get 297000000
Sep 21 12:27:42 rock-5b kernel: rockchip-hdptx-phy-hdmi fed60000.hdmiphy: hdptx_ropll_cmn_config bus_width:42625a00 rate:742500
Sep 21 12:27:42 rock-5b kernel: rockchip-hdptx-phy-hdmi fed60000.hdmiphy: hdptx phy pll locked!
Sep 21 12:27:42 rock-5b kernel: rockchip-hdptx-phy-hdmi fed60000.hdmiphy: hdptx_lcpll_cmn_config rate:40000000
Sep 21 12:27:42 rock-5b kernel: rockchip-hdptx-phy-hdmi fed60000.hdmiphy: hdptx phy pll locked!
Sep 21 12:27:42 rock-5b kernel: dwhdmi-rockchip fde80000.hdmi: final tmdsclk = 594000000
Sep 21 12:27:42 rock-5b kernel: dwhdmi-rockchip fde80000.hdmi: don't use dsc mode
Sep 21 12:27:42 rock-5b kernel: rockchip-hdptx-phy-hdmi fed60000.hdmiphy: bus_width:0x42625a00,bit_rate:40000000
Sep 21 12:27:42 rock-5b kernel: rockchip-hdptx-phy-hdmi fed60000.hdmiphy: hdptx phy lane locked!
Sep 21 12:27:43 rock-5b kernel: dwhdmi-rockchip fde80000.hdmi: goto ltsp
Sep 21 12:27:43 rock-5b kernel: dwhdmi-rockchip fde80000.hdmi: flt success
Sep 21 12:27:43 rock-5b /usr/libexec/gdm-x-session[2005]: (II) modeset(0): EDID vendor "SAM", prod id 29222
Sep 21 12:27:43 rock-5b /usr/libexec/gdm-x-session[2005]: (II) modeset(0): Using hsync ranges from config file
Sep 21 12:27:43 rock-5b /usr/libexec/gdm-x-session[2005]: (II) modeset(0): Using vrefresh ranges from config file
Sep 21 12:27:43 rock-5b /usr/libexec/gdm-x-session[2005]: (II) modeset(0): Printing DDC gathered Modelines:
Sep 21 12:27:43 rock-5b /usr/libexec/gdm-x-session[2005]: (II) modeset(0): Modeline "3840x2160"x0.0  594.00  3840 4016 4104 4400  2160 2168 2178 2250 +hsync +vsync (135.0 kHz eP)
...
Sep 21 12:27:43 rock-5b /usr/libexec/gdm-x-session[2005]: (--) modeset(0): HDMI max TMDS frequency 340000KHz
Sep 21 12:27:43 rock-5b kernel: rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_disable] Crtc atomic disable vp0
Sep 21 12:27:43 rock-5b kernel: rockchip-vop2 fdd90000.vop: [drm:vop2_isr] *ERROR* POST_BUF_EMPTY irq err at vp0
Sep 21 12:27:43 rock-5b kernel: rockchip-vop2 fdd90000.vop: [drm:vop2_isr] *ERROR* POST_BUF_EMPTY irq err at vp0
Sep 21 12:27:43 rock-5b kernel: rockchip-vop2 fdd90000.vop: [drm:vop2_isr] *ERROR* POST_BUF_EMPTY irq err at vp0
Sep 21 12:27:43 rock-5b kernel: rockchip-vop2 fdd90000.vop: [drm:vop2_isr] *ERROR* POST_BUF_EMPTY irq err at vp0
Sep 21 12:27:43 rock-5b kernel: rockchip-vop2 fdd90000.vop: [drm:vop2_isr] *ERROR* POST_BUF_EMPTY irq err at vp0
Sep 21 12:27:43 rock-5b kernel: rockchip-vop2 fdd90000.vop: [drm:vop2_isr] *ERROR* POST_BUF_EMPTY irq err at vp0
Sep 21 12:27:43 rock-5b kernel: rockchip-vop2 fdd90000.vop: [drm:vop2_isr] *ERROR* POST_BUF_EMPTY irq err at vp0
Sep 21 12:27:43 rock-5b kernel: rockchip-vop2 fdd90000.vop: [drm:vop2_isr] *ERROR* POST_BUF_EMPTY irq err at vp0
Sep 21 12:27:43 rock-5b kernel: rockchip-vop2 fdd90000.vop: [drm:vop2_isr] *ERROR* POST_BUF_EMPTY irq err at vp0
Sep 21 12:27:43 rock-5b kernel: rockchip-vop2 fdd90000.vop: [drm:vop2_isr] *ERROR* POST_BUF_EMPTY irq err at vp0
Sep 21 12:27:43 rock-5b /usr/libexec/gdm-x-session[2005]: (II) modeset(0): Allocate new frame buffer 3840x2160 stride
...`

Any suggestion welcome. Thanks!

Joshua-Riek, 2023-09-21:
I know there are EDID issues with a lot of monitors and displays. Maybe the kernel can not parse the EDID information for your 8k display properly? Also, have you tested this on Wayland?

zjeng, 2023-09-21:
We tested on Wayland, same story. 
The Modelines plot in the log shows **NOT** any 8k entry, indeed. However, xrandr list all 8k modi (refresh rates) as expected.
We have also added a dedicated timing to xrandr we know the monitor/cable work with from another source.
In the log we see the correct/expected pixel-clk (dclk)
```
> Sep 21 16:44:36 crt-desktop kernel: rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_enable] Update mode to 7680x4320p30, type: 11(if:800, flag:0x0) for vp0 dclk: **1186810000**
> Sep 21 16:44:37 crt-desktop kernel: rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_enable] Update mode to 3840x2160p60, type: 11(if:800, flag:0x0) for vp0 dclk: 594000000
```
I'll have a look at the EDID data later.

zjeng, 2023-09-21:
We have checked the EDID data under /sys/device/.../edid. Thes show the expected values BUT not in the _Detailed Timing Descriptors_ but in the _CTA-861 Extension Block_.

Might it be that the kernel version used is not able to parse VIC > 193?

```
$ edid-decode /sys/devices/platform/display-subsystem/drm/card0/card0-HDMI-A-1/edid
edid-decode (hex):

00 ff ff ff ff ff ff 00 4c 2d 26 72 00 0e 00 01
01 20 01 03 80 8e 50 78 0a b3 61 ab 4f 46 a8 27
0b 50 54 bd ef 80 71 4f 81 c0 81 00 81 80 95 00
a9 c0 b3 00 d1 c0 08 e8 00 30 f2 70 5a 80 b0 58
8a 00 50 1d 74 00 00 1e 56 5e 00 a0 a0 a0 29 50
30 20 35 00 50 1d 74 00 00 1a 00 00 00 fd 00 18
78 0f ff ee 00 0a 20 20 20 20 20 20 00 00 00 fc
00 53 41 4d 53 55 4e 47 0a 20 20 20 20 20 01 8c

02 03 6a f0 5f 61 60 65 66 75 76 da db c2 c3 c4
c6 c7 10 1f 04 13 05 14 20 21 22 5d 5e 5f 62 64
3f 40 03 12 2f 0d 57 07 09 07 07 15 07 50 57 07
01 67 54 03 83 0f 00 00 e2 00 4f e3 05 c3 01 6e
03 0c 00 10 00 b8 44 28 00 80 01 02 03 04 6d d8
5d c4 01 78 80 5b 02 00 00 c1 34 0b e3 06 0d 01
e3 0f ff 1f e5 01 8b 84 90 01 6f c2 00 a0 a0 a0
55 50 30 20 35 00 50 1d 74 00 00 1a 00 00 00 a7

----------------

Block 0, Base EDID:
  EDID Structure Version & Revision: 1.3
  Vendor & Product Identification:
    Manufacturer: SAM
    Model: 29222
    Serial Number: 16780800
    Made in: week 1 of 2022
  Basic Display Parameters & Features:
    Digital display
    Maximum image size: 142 cm x 80 cm
    Gamma: 2.20
    RGB color display
    First detailed timing is the preferred timing
  Color Characteristics:
    Red  : 0.6699, 0.3115
    Green: 0.2734, 0.6591
    Blue : 0.1533, 0.0449
    White: 0.3125, 0.3291
  Established Timings I & II:
    IBM     :   720x400    70.082 Hz   9:5    31.467 kHz  28.320 MHz
    DMT 0x04:   640x480    59.940 Hz   4:3    31.469 kHz  25.175 MHz
    Apple   :   640x480    66.667 Hz   4:3    35.000 kHz  30.240 MHz
    DMT 0x05:   640x480    72.809 Hz   4:3    37.861 kHz  31.500 MHz
    DMT 0x06:   640x480    75.000 Hz   4:3    37.500 kHz  31.500 MHz
    DMT 0x09:   800x600    60.317 Hz   4:3    37.879 kHz  40.000 MHz
    DMT 0x0a:   800x600    72.188 Hz   4:3    48.077 kHz  50.000 MHz
    DMT 0x0b:   800x600    75.000 Hz   4:3    46.875 kHz  49.500 MHz
    Apple   :   832x624    74.551 Hz   4:3    49.726 kHz  57.284 MHz
    DMT 0x10:  1024x768    60.004 Hz   4:3    48.363 kHz  65.000 MHz
    DMT 0x11:  1024x768    70.069 Hz   4:3    56.476 kHz  75.000 MHz
    DMT 0x12:  1024x768    75.029 Hz   4:3    60.023 kHz  78.750 MHz
    DMT 0x24:  1280x1024   75.025 Hz   5:4    79.976 kHz 135.000 MHz
    Apple   :  1152x870    75.062 Hz 192:145  68.681 kHz 100.000 MHz
  Standard Timings:
    DMT 0x15:  1152x864    75.000 Hz   4:3    67.500 kHz 108.000 MHz
    DMT 0x55:  1280x720    60.000 Hz  16:9    45.000 kHz  74.250 MHz
    DMT 0x1c:  1280x800    59.810 Hz  16:10   49.702 kHz  83.500 MHz
    DMT 0x23:  1280x1024   60.020 Hz   5:4    63.981 kHz 108.000 MHz
    DMT 0x2f:  1440x900    59.887 Hz  16:10   55.935 kHz 106.500 MHz
    DMT 0x53:  1600x900    60.000 Hz  16:9    60.000 kHz 108.000 MHz (RB)
    DMT 0x3a:  1680x1050   59.954 Hz  16:10   65.290 kHz 146.250 MHz
    DMT 0x52:  1920x1080   60.000 Hz  16:9    67.500 kHz 148.500 MHz
[... 129 lines cut ...]
    Supported static metadata descriptors:
      Static metadata type 1
  YCbCr 4:2:0 Capability Map Data Block:
    VIC  97:  3840x2160   60.000 Hz  16:9    67.500 kHz 297.000 MHz
    VIC  96:  3840x2160   50.000 Hz  16:9    56.250 kHz 297.000 MHz
    VIC 101:  4096x2160   50.000 Hz 256:135  56.250 kHz 297.000 MHz
    VIC 102:  4096x2160   60.000 Hz 256:135  67.500 kHz 297.000 MHz
    VIC 117:  3840x2160  100.000 Hz  16:9   112.500 kHz 594.000 MHz
    VIC 118:  3840x2160  120.000 Hz  16:9   135.000 kHz 594.000 MHz
    VIC 218:  4096x2160  100.000 Hz 256:135 112.500 kHz 594.000 MHz
    VIC 219:  4096x2160  120.000 Hz 256:135 135.000 kHz 594.000 MHz
    VIC 194:  7680x4320   24.000 Hz  16:9    54.000 kHz 594.000 MHz
    VIC 195:  7680x4320   25.000 Hz  16:9    55.000 kHz 594.000 MHz
    VIC 196:  7680x4320   30.000 Hz  16:9    66.000 kHz 594.000 MHz
    VIC 198:  7680x4320   50.000 Hz  16:9   110.000 kHz 1188.000 MHz
    VIC 199:  7680x4320   60.000 Hz  16:9   132.000 kHz 1188.000 MHz
  Vendor-Specific Video Data Block (HDR10+), OUI 90-84-8B:
    Application Version: 1
  Detailed Timing Descriptors:
    DTD 3:  2560x1440  119.998 Hz  16:9   182.996 kHz 497.750 MHz (1872 mm x 1053 mm)
                 Hfront   48 Hsync  32 Hback  80 Hpol P
                 Vfront    3 Vsync   5 Vback  77 Vpol N
Checksum: 0xa7

```

zjeng, 2023-09-22:
I just spotted that the kernel log has a problem with 8kp60
```
Sep 22 14:21:43 crt-desktop kernel: rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_enable] Update mode to 7680x4320p60, type: 11(if:800, flag:0x0) for vp0 dclk: -1918967296
Sep 22 14:21:29 crt-desktop kernel: rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_enable] Update mode to 3840x2160p60, type: 11(if:800, flag:0x0) for vp0 dclk: 594000000
Sep 22 14:21:28 crt-desktop kernel: rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_enable] Update mode to 7680x4320p30, type: 11(if:800, flag:0x0) for vp0 dclk: 1188000000
```
The reported dclk frequency is close to 32b integer range limit for 8kp30 but exceed this range for 8kp60. Thus, a negativ (turned over) number is reported.
Seeing this, I wonder if there are other range violations in the kernel which lead to misconfiguration rather than just to wrong reporting.

closed 2025-02-17 by github-actions[bot]