Joshua-Riek/ubuntu-rockchip #374: Input Device - HDMI-In Audio does not exist on v1.27

guynich, 2023-09-23:
OrangePi 5 Plus: I do not see input HDMI-IN audio in the Settings > Sound > Input Device dropdown on v1.27 Desktop.

Potential duplicate of #243.   I did try running `sudo alsa force-reload` but did not fix.

Joshua-Riek, 2023-09-23:
Yes, this is a duplicate. The HDMI in audio device does exist, but the settings interface does not show it. 

I have marked this off as a very low priority because the audio device is still accessible.

guynich, 2023-09-25:
Thanks Joshua

When I connect to the OrangePi 5 Plus HDMI-in (from a MacBook: it shows `RK-UHD` as the Output sound device) I do not see the input audio device HDMI-In using `pulseaudio` from the command line, e.g.:
```
orangepi@orangepi-plus:~$ pacmd list-sources | grep -e 'name:'
	name: <alsa_output.platform-dp0-sound.stereo-fallback.monitor>
	name: <alsa_output.platform-es8388-sound.stereo-fallback.monitor>
	name: <alsa_output.platform-hdmi0-sound.stereo-fallback.monitor>
	name: <alsa_output.platform-hdmi1-sound.stereo-fallback.monitor>
orangepi@orangepi-plus:~$ 
```
I'm wondering how the audio device is accessible if not through pulseaudio.

guynich, 2023-09-27:
Some more info: I tried OKdo Rock 5B with release 1.27 Desktop e.g.: `ubuntu-22.04.3-preinstalled-desktop-arm64-rock-5b.img.xz`  and HDMI input is seen as `Analog Input - HDMI-In Audio` in the Settings > Sound > Input Device dropdown.  That's great!

guynich, 2023-09-30:
Closing as duplicate.

closed 2023-09-30 by guynich