Joshua-Riek/ubuntu-rockchip #375: OBS Crashes

za-songguo, 2023-09-24:
When I record video using OBS (gstreamer hardware encoding) and OBS randomly crashes when I click the OBS icon in the dock.
Here's the log:
```
info: ==== Recording Start ===============================================
info: [ffmpeg muxer: 'adv_file_output'] Writing file '/home/zasongguo/2023-09-24 13-03-55.mp4'...
info: output adv_file_output paused
info: output adv_file_output unpaused
QObject::startTimer: Timers cannot have negative intervals
info: Pausing recording due to hotkey
info: output adv_file_output paused
info: Unpausing recording due to hotkey
info: output adv_file_output unpaused
QObject::startTimer: Timers cannot have negative intervals
The Wayland connection broke. Did the Wayland compositor die?
```

za-songguo, 2023-10-01:
I have found a solution to this problem.
Follow the steps [here](https://github.com/Joshua-Riek/ubuntu-rockchip/discussions/28) to install OBS Studio first. 
We need to install the Qt XCB plugin and `qt6-wayland`:
```
sudo apt install qt6-qpa-plugins xcb qt6-wayland
```
And we should edit `/etc/environment` and replace `QT_QPA_PLATFORM=wayland` with `QT_QPA_PLATFORM="xcb;wayland"` and run `source /etc/environment`.
After editing, we should see `info: Window System: X11.0, Vendor: The X.Org Foundation, Version: 1.22.1` in the OBS Studio's startup log(use a terminal to run OBS Studio using the `obs` command).
The OBS installation guide has a problem, I think there's a Qt bug that causes us to not be able to run OBS Studio on Wayland, so we should use XWayland to run it.
(For more information: [OBS and Armbian ](https://forum.radxa.com/t/obs-and-armbian/15668/15))

Joshua-Riek, 2023-10-01:
Good find, it looks like the OBS package does not have `qt6-qpa-plugins`, `xcb`, and `qt6-wayland` listed as dependencies which would cause this issue.

za-songguo, 2023-10-02:
> Good find, it looks like the OBS package does not have `qt6-qpa-plugins`, `xcb`, and `qt6-wayland` listed as dependencies which would cause this issue.

The key to solving this problem is to run OBS on XCB, using XWayland instead of Wayland.

closed 2023-10-02 by za-songguo

za-songguo, 2023-10-02:
> I have found a solution to this problem. Follow the steps [here](https://github.com/Joshua-Riek/ubuntu-rockchip/discussions/28) to install OBS Studio first. We need to install the Qt XCB plugin and [...]

Remember to run OBS in a terminal because the environment variable `QT_QPA_PLATFORM` does not seem to have any effect when OBS is started from the desktop.

stallion25, 2024-03-04:
Adding qt6 wayland fixed problem.

stallion25, 2024-03-04:
I have problem with HDMIR on this version. Cheese is working fine after reboot. After any try to touch HDMIRX from 412 tester or obs is crashing it. Cheese works no more. need to reboot. I think this is serious release issue. Cany anyone with HDMIRX with example OPI5+ or ROCK5B can confirm this ?

MaximKurakin, 2024-06-13:
> Cany anyone with HDMIRX with example OPI5+ or ROCK5B can confirm this ?

Oh yeah!  I have Orange PI 5 PLUS and have the same issue. Welcome to the suffering club ! https://github.com/Joshua-Riek/ubuntu-rockchip/issues/830.