Joshua-Riek/ubuntu-rockchip #376: Failed to start Wait for Network to be Configured

Zaanza, 2023-09-24:
Hello, after updating Ubuntu Server on my Orange Pi 5, I have to wait two minutes on eachboot, I have this error :

A start job is running for Wait for Network to be Configured.

2 minutes later :
Failed to start Wait for Network to be Configured
See 'systemctl status system-networkd-wait-online.service' for details.

The command send me back
Unit system-networkd-wait-online.service could not be found.

I'm afraid I'll lose ssh access if I try something, can you help me ? THANKS !

Joshua-Riek, 2023-09-24:
I recently came up with a workaround for this https://github.com/Joshua-Riek/ubuntu-rockchip/commit/53d348049c41b01a78f325473c16de1c135a79c3. It will set the timeout for the service to 10 seconds instead of 120. 

I have a build with this fix here: https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/6284505659

You can modify the systemd service like below:

```
sudo systemctl edit --full systemd-networkd-wait-online.service
```
And edit the command to:
```
ExecStart=/usr/lib/systemd/systemd-networkd-wait-online --timeout=10
```

closed 2023-09-24 by Joshua-Riek

Zaanza, 2023-09-24:
Thank you, I will try, but is there no solution to fix the problem at the source ?

Joshua-Riek, 2023-09-24:
This issue has been in generic Ubuntu server images for a long time. I've seen reports from Ubuntu 18.04 with this issue.

Zaanza, 2023-09-24:
It seems to work, however since I have this error I also have a similar error via the discord.js api with node (timeout connection), I think that the --timeout will not help me with that, thank you anyway.