Joshua-Riek/ubuntu-rockchip #378: Orange Pi 5 Plus w/ official OPi WiFi/BT card - WiFi does not work with Bluetooth enabled

coolva, 2023-09-25:
I've been having an issue where I restart the device and when it reboots, bluetooth is always checked enabled.  I have to go into the settings, disable bluetooth and reboot before WiFi works each time.

Deiiji, 2023-09-26:
This is a known issue, I have stop bt in shutdown script and start it with a delay in startup, waiting for a  new driver.

Joshua-Riek, 2023-09-26:
I could not reproduce this issue, but I don't have the official WiFi + BT card, but it's the same model. I will need to order the official one, which will take one or two weeks to deliver.

jcochran, 2023-09-26:
I am also having an issue with the Orange Pi 5 Plus and the RTL8852be WiFi+Bluetooth. During install and after a couple of initial reboots the WiFi works correctly, but then eventually stops working on boot. I have found that on reboot/power up, I need to remove and insert the 8852be kernel module:

sudo modprobe -r 8852be
sudo modprobe -i 8852be

If I do that, then WiFi works correctly during that session. The Orange Pi 5+ is running your 1.27 release. Prior to the 1.27 release, the behavior was different in that the remove/insert did not always work. Since applying all updates to the older release and then reinstalling cleaning with the 1.27 due to another issue, the remove/insert has been reliable to resolving the WiFi issue.

I did try blacklisting 8852be during boot and the simply inserting 8852be later at login manually, but found regardless the module seems to need to be inserted twice for it to work.

In the dmesg output all the related messages are the same as previously documented.

MarkhamLee, 2023-09-26:
I've experienced the same issue, with a couple of extra wrinkles: 

* I bought an Intel AX-210 card, put that in the device, it didn't work 
* Bought the official card and it worked fine, but Bluetooth didn't work. However, Wi-Fi was kinda slow 40 MB/S 
* Setup the device to boot from NVME, rebooted, updated, Wi-Fi worked at first, but stopped working on subsequent reboots, Bluetooth didn't work either. 
* Put back in the AX-210 card and now Wi-Fi and Bluetooth work just fine. Plus, Wi-Fi is fast:  270 MB/s 

So far, haven't had any issues with the AX-210, will probably just stick with that.

Joshua-Riek, 2023-09-26:
> I am also having an issue with the Orange Pi 5 Plus and the RTL8852be WiFi+Bluetooth. During install and after a couple of initial reboots the WiFi works correctly, but then eventually stops working on boot. I have found that on reboot/power up, I need to remove and insert the 8852be kernel module:
> 
> sudo modprobe -r 8852be sudo modprobe -i 8852be
> 
> If I do that, then WiFi works correctly during that session. The Orange Pi 5+ is running your 1.27 release. Prior to the 1.27 release, the behavior was different in that the remove/insert did not always work. Since applying all updates to the older release and then reinstalling cleanly with the 1.27 due to another issue, the remove/insert has been reliable to resolving the WiFi issue.
> 
> I did try blacklisting 8852be during boot and the simply inserting 8852be later at login manually, but found regardless the module seems to need to be inserted twice for it to work.
> 
> In the dmesg output all the related messages are the same as previously documented.

This might be an appropriate workaround for now. Later today I can create a system service to remove and insert the 8852be module on boot. I will post a link to the build download for you guys to test with when it's done.

Joshua-Riek, 2023-09-26:
> I've experienced the same issue, with a couple of extra wrinkles:
>
>     * I bought an Intel AX-210 card, put that in the device, it didn't work
>
>     * Bought the official card and it worked fine, but [...]

I noticed some users having issues with AX-210, but could not reproduce. I personally use AX-210 in my boards, it's been the most stable from my experience.

coolva, 2023-09-26:
Thanks everyone!  Let me know if I can help with this in any way...

Joshua-Riek, 2023-09-26:
I started a build below, and it should be done in 2-3 hours. I don't have the official WiFi + BT card, so could you test the Orange Pi 5 Plus image once it's done? That would be great.

https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/6319293461

Aashitech, 2023-09-26:
> I started a build below, and it should be done in 2-3 hours. I don't have the official WiFi + BT card, so could you test the Orange Pi 5 Plus image once it's done? That would be great.
> 
> https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/6319293461

I can arrange you that OPI-5Plus official WIFI+BT module if you wish , please let me know .

Joshua-Riek, 2023-09-26:
> I can arrange you that OPI-5Plus official WIFI+BT module if you wish , please let me know .

I have one ordered, but shipping is slow, expected to be delivered in 1-2 weeks from today.

Aashitech, 2023-09-27:
> I have one ordered, but shipping is slow, expected to be delivered in 1-2 weeks from today.

I would have arranged much faster... If you don`t mind then can I have your email ? there are few other things to discuss with you which can`t be discussed in public for now .

Joshua-Riek, 2023-09-27:
> I would have arranged much faster... If you don`t mind then can I have your email ? there are few other things to discuss with you which can`t be discussed in public for now .

Sure, you can reach me at jjriek@verizon.net.

jcochran, 2023-09-27:
I applied the committed service file to my OPI 5+ and did multiple reboots and power cycles and the WiFi was present every single time at the GDM login screen without any user interaction.

Joshua-Riek, 2023-09-27:
> I applied the committed service file to my OPI 5+ and did multiple reboots and power cycles and the WiFi was present every single time at the GDM login screen without any user interaction.

Hopefully, I can come up with a more permanent fix in the future. But its great to hear its working!

Joshua-Riek, 2023-09-28:
> I started a build below, and it should be done in 2-3 hours. I don't have the official WiFi + BT card, so could you test the Orange Pi 5 Plus image once it's done? That would be great.
> 
> https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/6319293461

Anyone able to test this?

coolva, 2023-10-01:
I did the same as @jcochran and have had no issues since.  I didn't want to re-image my drive due to spending so long getting everything setup like I wanted.

Joshua-Riek, 2023-10-01:
Got it thanks for the feedback! Ill close this for now as this seems like a valid workaround.

closed 2023-10-01 by Joshua-Riek