Joshua-Riek/ubuntu-rockchip #380: (orangpi5-plus)help!,idon't known how to enable serial or can

2350015226, 2023-09-26:
In the official tutorial, orangepi-config is used to enable the serial port and can.but i can't. 
[image]
[image]

Joshua-Riek, 2023-09-26:
This is not an official distro from Orange Pi, and it does not have the orangepi-config program. However, you can still enable UART and CAN interfaces by using device tree overlays. You must specify the overlays you wish to enable as shown on the wiki page below.

https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Device-Tree-Overlays

2350015226, 2023-09-27:
> This is not an official distro from Orange Pi, and it does not have the orangepi-config program. However, you can still enable UART and CAN interfaces by using device tree overlays. You must specify [...]
I have tried your method, and now I can see my enabled device on the console, thank you. If you have the chance to come to Guangdong Province, I will definitely take you to eat the freshest seafood, haha, thanks again

closed 2023-09-27 by Joshua-Riek