Joshua-Riek/ubuntu-rockchip #381: rock 5b  3.5mm jack issue

hemsinghb, 2023-09-26:
Hi Sir,
 a.   3.5 mm audio jack works only for audio out (speakers) only,  the mic is not enabled / not working. 
 b.   No GPIO library, So please introduce GPIO on Rock 5b.
 c.  OLED not able fix on rock 5b... no library 
    
    Hope and request you for the solutions for the above, please.

Joshua-Riek, 2023-09-26:
a. I'm aware of the issue but have not been able to look into it yet.
b. There is no way I'm going to create an entire GPIO library.
c. I don't know what this means.

hemsinghb, 2023-09-27:
Thanks for response.  Item ( c ) is regarding OLED  ( generally 0.96 / 1.3" LED small display  connected to i2c port) . This is also a library issue, similar like GPIO library. 
...with regards.

closed 2023-09-27 by hemsinghb