Joshua-Riek/ubuntu-rockchip #386: sata boot

digby5, 2023-09-28:
i searched and found a thread for sata boot using fdt_overlays=rk3588-ssd-sata.dtbo. i put that in ubunutenv and no luck. created a new overlays.txt and no luck.  that dtbo file is actually missing in the boot directories. im using the latest build. is sata boot disabled for the latest build or is there another method?

Joshua-Riek, 2023-09-28:
I need some more context here, what board?

Joshua-Riek, 2023-09-28:
Ah remove the `.dtbo` in the overlay name.

digby5, 2023-09-29:
its opi5. this will work even tho'  there is no such file in the directories?

Joshua-Riek, 2023-09-29:
Use the overlay `orangepi-5-sata`

Joshua-Riek, 2023-09-29:
Then flash the bootloader with a sata bootloader `sudo dd if=/lib/u-boot-orangepi-rk3588/rkspi_loader_sata.img of=/dev/mtdblock0 conv=notrunc `

digby5, 2023-09-29:
it worked. much smoother than armbian bookworm since they broke gpu accel. i just noticed this build can run geekbench -gpu; before on all the other builds, geekbench would complain and never allow u to run the gpu bench. are u still using the ppa gpu drivers or is this something else?

long shot but u wouldnt know how to boot sata on the official orange pi os arch? i tried this [https://forum.armbian.com/topic/26136-guide-how-to-boot-orange-pi-5-from-sata-m2-ssd-for-orange-pi-os-and-armbian/](url) guide with overlays=ssd-sata and no dice - there isnt even a txt file in the boot directory. someone mentioned it supported vulkan - not sure if they mixed it up with the droid edition, so wanted to see if its true.

Joshua-Riek, 2023-10-01:
Good to hear this is working, I'm using amazingfates ppa for gpu drivers. Linux does not support Vulkan at all, you might have some conflicting info, I don't know how the orange pi os arch works, so I can not help.

closed 2023-10-01 by Joshua-Riek

digby5, 2023-10-02:
jetson has vulkan support and that is arm linux. the arch repo has a vulkan rockchip package but i couldn't get it to work.