Joshua-Riek/ubuntu-rockchip #391: No image on 4k TV (1 second purple color, then black screen)

InfernalWolf666, 2023-10-01:
I'm using Google Translate, sorry for any mistakes.

Hello my dear friend!
I've encountered a problem!
Let's start with the fact that I'm a complete noob in the world of Linux.

When I install on my orange pi 5 plus on EmmC 256 GB from the official website (http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/Orange-Pi-emmc.html), operating systems:

1.Pi OS droid
2.Pi Os arch
3.Ubuntu Pi
4.Debian Pi

http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/service-and-support/Orange-Pi-5-plus.html

alternative OS:
1. OpenFyde OS
https://openfyde.io/

2.FydeOS
https://fydeos.io/

All of these systems output images to my 4K TV.

And everyone has the official Bluetooth and Wi-Fi module

http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/Orange-Pi-R6.html

And all other operating systems that are not from the manufacturer do not display images on my TV screen.

It looks like this, I switch to the HDMI channel for the first second I see the Ubuntu logo in purple, then a black screen.

I take your Ubuntu operating system and go to my 720p plasma TV.

I connect and the image is displayed (but unfortunately the image goes off the screen, and the required image resolution of 1024 x 768 is missing)

Therefore, tell me how to fix the display for a 4K TV?

And how to add screen resolution for a plasma TV 1024x768?

lukaszsobala, 2023-10-11:
I have a similar problem. My 1024x600 touchscreen does not work. I tried the ubuntu image and Armbian (changing different options in ArmbianEnv.txt).

I can provide more info. The display only works with kernel 5.10.110, it does not work with any systems with 5.10.160 on Rock 5B (but it works with this kernel on Rock 5A).

Joshua-Riek, 2023-10-28:
There are EDID issues with the kernel, this will not be fixed.