Joshua-Riek/ubuntu-rockchip #392: Support for Firefly ROC-RK3588S

thesleort, 2023-10-01:
I stumbled upon this project and I am interested to know how easy/hard it is to add support for Firefly's ROC-RK3588S. It is an SBC similar to the ones listed for this project. They have some documentation, but their supported Ubuntu is 18.04, which I think is quite old. 

If I can I would like to help, but it is not my main expertise getting SBCs to boot. I also cannot find much information in the wiki on how to add new devices.

https://wiki.t-firefly.com/en/ROC-RK3588S-PC/index.html

Joshua-Riek, 2023-10-01:
It should not be too hard to add a new device. Take a look at this commit as a reference https://github.com/Joshua-Riek/ubuntu-rockchip/commit/3d473b15bc7c0a4c663202913fdd0a91dab70b73. The only issue is my custom flash-kernel package, i would need the exact device tree model so the system can update the kernel, initrd, and dtb to the boot partition.

However, I do need to be careful about what boards are added. I have limited hardware resources for building, and the build system was never meant to handle more than a few boards.

thesleort, 2023-10-02:
Thank you. I will try and look into this.

I made some digging and found that it is possibly already part of your Linux fork. https://github.com/Joshua-Riek/linux-rockchip/blob/linux-5.10-gen-rkr5.1/arch/arm64/boot/dts/rockchip/rk3588-evb1-lp4-v10-linux.dts

I base that on this finding in one of Firefly's repositories for building Ubuntu for this SBC. https://gitlab.com/firefly-linux/device/rockchip/-/blob/firefly/rk3588/boot.its?ref_type=heads

thesleort, 2023-10-02:
I am able to build the mainline linux kernel using `sudo ./build.sh -m --board=firefly` which points to the `rk3588-evb1-lp4-v10-linux.dtb`, that I think Firefly uses.

However, once it reaches the uboot section it fails, since I haven't made anything for that part yet:
```
find: ‘build/u-boot-firefly_*.deb’: No such file or directory
./scripts/build-u-boot.sh: line 21: ../packages/u-boot-firefly-rk3588/debian/upstream: No such file or directory
Error: in ./build.sh on line 152
```
Is there a smart way to create this similar to the others in the `packages/` directory?

My changes are found in https://github.com/thesleort/ubuntu-rockchip/tree/firefly-rk3588s-support

closed 2023-10-28 by Joshua-Riek