Joshua-Riek/ubuntu-rockchip #393: KVM support

JeffersonTerrerfield, 2023-10-02:
Hey,

first of all, I must say that you have created a great project. Thank you so much!

I don't know a lot about virtualization since I haven't used it much in the past. However, I wanted to try it out on my rk3588 board.

As far as I know qemu support arm64 hosts, but qemu alone is pretty slow. To achieve near native speed, one has to use qemu with kvm. KVM also supports arm64. However, it seems the kernel was compiled without kvm support.

I guess the rk3588 has enough power for multiple VMs. I hope you consider adding support for it in the near future.

Once again, thank you for your efforts.

Joshua-Riek, 2023-10-02:
Where do you see that the kernel is compiled without KVM support? I see it in the config.
```
joshua@GS66-Stealth-11UE:~/Downloads/linux-rockchip$ cat .config | grep -i kvm
CONFIG_KVM=y
CONFIG_HAVE_KVM_IRQCHIP=y
CONFIG_HAVE_KVM_IRQFD=y
CONFIG_HAVE_KVM_IRQ_ROUTING=y
CONFIG_HAVE_KVM_EVENTFD=y
CONFIG_KVM_MMIO=y
CONFIG_HAVE_KVM_MSI=y
CONFIG_HAVE_KVM_CPU_RELAX_INTERCEPT=y
CONFIG_KVM_VFIO=y
CONFIG_HAVE_KVM_ARCH_TLB_FLUSH_ALL=y
CONFIG_KVM_GENERIC_DIRTYLOG_READ_PROTECT=y
CONFIG_HAVE_KVM_IRQ_BYPASS=y
CONFIG_HAVE_KVM_VCPU_RUN_PID_CHANGE=y
```

JeffersonTerrerfield, 2023-10-04:
Thank you very much for your answer.

You are right. It seems, that the Kernel supports KVM correctly. The problem must be somewhere else.

Unfortunately, I can't try it out again, since I currently have no time. When I try it again in the near future and run into issues related to this project, I will reopen the issue.

closed 2023-10-04 by JeffersonTerrerfield