Joshua-Riek/ubuntu-rockchip #394: ZFS Support

joshdick, 2023-10-02:
tl;dr: It seems like `ubuntu-rockchip`'s kernel customizations are incompatible with the ZFS kernel module; is there any possibility that compatibility could be added? Or is there already some other known way to get ZFS working?

Thanks for all of the work you do on `ubuntu-rockchip`!

---

I am attempting to use ZFS on an OrangePi 5 running the v1.27 `ubuntu-22.04.3-preinstalled-server-arm64-orangepi-5.img.xz` image.

Attempting to `sudo apt install zfsutils-linux` results in:

```
Setting up libnvpair3linux (2.1.5-1ubuntu6~22.04.1) ...
Setting up libuutil3linux (2.1.5-1ubuntu6~22.04.1) ...
Setting up libzfs4linux (2.1.5-1ubuntu6~22.04.1) ...
Setting up libzpool5linux (2.1.5-1ubuntu6~22.04.1) ...
Setting up zfsutils-linux (2.1.5-1ubuntu6~22.04.1) ...
modprobe: FATAL: Module zfs not found in directory /lib/modules/5.10.160-rockchip
Created symlink /etc/systemd/system/zfs-import.target.wants/zfs-import-cache.service → /lib/systemd/system/zfs-import-cache.service.
Created symlink /etc/systemd/system/zfs.target.wants/zfs-import.target → /lib/systemd/system/zfs-import.target.
Created symlink /etc/systemd/system/zfs-mount.service.wants/zfs-load-module.service → /lib/systemd/system/zfs-load-module.service.
Created symlink /etc/systemd/system/zfs.target.wants/zfs-load-module.service → /lib/systemd/system/zfs-load-module.service.
Created symlink /etc/systemd/system/zfs.target.wants/zfs-mount.service → /lib/systemd/system/zfs-mount.service.
Created symlink /etc/systemd/system/zfs.target.wants/zfs-share.service → /lib/systemd/system/zfs-share.service.
Created symlink /etc/systemd/system/zfs-volumes.target.wants/zfs-volume-wait.service → /lib/systemd/system/zfs-volume-wait.service.
Created symlink /etc/systemd/system/zfs.target.wants/zfs-volumes.target → /lib/systemd/system/zfs-volumes.target.
Created symlink /etc/systemd/system/multi-user.target.wants/zfs.target → /lib/systemd/system/zfs.target.
zfs-import-scan.service is a disabled or a static unit, not starting it.
Setting up zfs-zed (2.1.5-1ubuntu6~22.04.1) ...
Created symlink /etc/systemd/system/zed.service → /lib/systemd/system/zfs-zed.service.
Created symlink /etc/systemd/system/zfs.target.wants/zfs-zed.service → /lib/systemd/system/zfs-zed.service.
Processing triggers for man-db (2.10.2-1) ...
Processing triggers for libc-bin (2.35-0ubuntu3.1+3) ...
ubuntu@ubuntu:~$ sudo modprobe zfs
modprobe: FATAL: Module zfs not found in directory /lib/modules/5.10.160-rockchip
```

And attempting to `sudo apt install zfs-dkms` results in:

```
Setting up zfs-dkms (2.1.5-1ubuntu6~22.04.1) ...
Loading new zfs-2.1.5 DKMS files...
Building for 5.10.160-rockchip
Building initial module for 5.10.160-rockchip
configure: error:
	*** None of the expected "file_fallocate" interfaces were detected.
	*** This may be because your kernel version is newer than what is
	*** supported, or you are using a patched custom kernel with
	*** incompatible modifications.
	***
	*** ZFS Version: zfs-2.1.5-1ubuntu6~22.04.1
	*** Compatible Kernels: 3.10 - 5.19

ERROR (dkms apport): kernel package linux-headers-5.10.160-rockchip is not supported
Error! Bad return status for module build on kernel: 5.10.160-rockchip (aarch64)
Consult /var/lib/dkms/zfs/2.1.5/build/make.log for more information.
dpkg: error processing package zfs-dkms (--configure):
 installed zfs-dkms package post-installation script subprocess returned error exit status 10
Processing triggers for libc-bin (2.35-0ubuntu3.1+3) ...
Processing triggers for man-db (2.10.2-1) ...
Processing triggers for initramfs-tools (0.140ubuntu13.4) ...
update-initramfs: Generating /boot/initrd.img-5.10.160-rockchip
flash-kernel: installing version 5.10.160-rockchip
Errors were encountered while processing:
 zfs-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Joshua-Riek, 2023-10-02:
The kernel is a mess. It's an android port from Rockchip. There are many weird things and bugs because of this. One could make an argument that this is not even kernel 5.10.160, but it's all we have that fully supports the hardware.

I'm not sure what the requirements are for zfs, but that file_fallocate error seems like a start for debugging this.

juliosene, 2023-10-02:
I'm also trying to get ZFS on Ubuntu 22.04. I tried to compile the open-zfs source code and the error was similar.

```
checking whether enum node_stat_item contains NR_FILE_PAGES... yes
checking whether enum node_stat_item contains NR_INACTIVE_ANON... yes
checking whether enum node_stat_item contains NR_INACTIVE_FILE... yes
checking whether enum zone_stat_item contains NR_FILE_PAGES... no
checking whether enum zone_stat_item contains NR_INACTIVE_ANON... no
checking whether enum zone_stat_item contains NR_INACTIVE_FILE... no
checking whether global_page_state enums are sane... yes
checking whether compile-time stack validation (objtool) is available... no
checking whether pde_data() is lowercase... no
checking whether fops->fallocate() exists... configure: error:
        *** None of the expected "file_fallocate" interfaces were detected.
        *** This may be because your kernel version is newer than what is
        *** supported, or you are using a patched custom kernel with
        *** incompatible modifications.
        ***
        *** ZFS Version: zfs-2.1.13-1
        *** Compatible Kernels: 3.10 - 6.5
```

The Ubuntu desktop version distributed by Orange Pi comes with zfs and works fine. (but I have different issues with this Ubuntu version...)

Joshua-Riek, 2023-10-02:
This looks relevant to zfs: https://github.com/orangepi-xunlong/orangepi-build/issues/82

joshdick, 2023-10-03:
> This looks relevant to zfs: [orangepi-xunlong/orangepi-build#82](https://github.com/orangepi-xunlong/orangepi-build/issues/82)

Thanks @Joshua-Riek for directing me to that issue, I've left [a comment](https://github.com/orangepi-xunlong/orangepi-build/issues/82#issuecomment-1744338706) on it!

juliosene, 2023-10-10:
Hello! This still a working in progres, but I've made some progress compiling the kernel with zfs. On the first attempt, I got the following error message. 

```
checking for kernel config option compatibility... done
checking whether CONFIG_MODULES is defined... no
configure: error:
                *** This kernel does not include the required loadable module
                *** support!
                ***
                *** To build OpenZFS as a loadable Linux kernel module
                *** enable loadable module support by setting
                *** `CONFIG_MODULES=y` in the kernel configuration and run
                *** `make modules_prepare` in the Linux source tree.
                ***
                *** If you don't intend to enable loadable kernel module
                *** support, please compile OpenZFS as a Linux kernel built-in.
                ***
                *** Prepare the Linux source tree by running `make prepare`,
                *** use the OpenZFS `--enable-linux-builtin` configure option,
                *** copy the OpenZFS sources into the Linux source tree using
                *** `./copy-builtin <linux source directory>`,
                *** set `CONFIG_ZFS=y` in the kernel configuration and compile
                *** kernel as usual.
```

So I adopted the following procedure:
- I downloaded kernel version 5.15.134 ([Linux Kernel](https://www.kernel.org/))
- Installed the dependencies for compilation
`sudo apt-get install libncurses-dev gawk flex bison openssl libssl-dev dkms libelf-dev libudev-dev libpci-dev libiberty-dev autoconf llvm`
- I downloaded the current version of the zfs source code ([ZFS source code](https://zfsonlinux.org/)) [some instructions](https://openzfs.github.io/openzfs-docs/Developer%20Resources/Building%20ZFS.html#installing-dependencies)
- Decompressed the kernel and zfs
- Ran the configure to setup zfs as an integral part of linux kernel.
`./configure --with-linux=/<YOUR PATH>/linux-5.15.134/ --enable-linux-builtin`
- Compiled zfs and the linux kernel. No errors.

Now I need to install and test

Joshua-Riek, 2023-10-10:
That kernel will not work with the RK3588 hardware. I have a work-in-progress fork of 6.6.0-rc5 that you would have better luck with—however, not all the hardware works, such as the GPU, NPU, and HDMI. You will need a serial console to access the device.

https://github.com/Joshua-Riek/linux/tree/v6.6-rc5-rk3588

TomKVM, 2023-10-16:
Hi, great to see that I not the only one with this particular challenge. I am currently trying to get ZFS working on a OrangePi 5 Plus with not luck. The Opi distros will work with their now version of DEB packages however they bring other bugs - in my case the 2.5gbe interfaces are underperforming (doing an iperf3 test upstream gets the correct speed but downstream is super slow transfer). I would love to have ZFS working on Joshua's Ubuntu's fork ! **I do have a question thou : Which Kernel header should be used if I wanted to use the ZFS deb packages provided on the OPi5 google drive ? Where should I get it from ? I have tried their DEB packages on Joshua's latest desktop ubuntu 22.04 for the Orangepi 5 Plus but no luck there.. -** THANK YOU Josh for all the great work you do here.. really love using your version of Ubuntu on my Orange Pi 5 plus !

Joshua-Riek, 2023-10-16:
Orangepi uses kernel 5.10.110, while I have images with 5.10.160. But you could try mainline Linux which is still under development and not all hardware is working, you would need a serial console to interact with the advice as HDMI does not work.

https://github.com/Joshua-Riek/ubuntu-rockchip/actions/workflows/build-mainline.yml

juliosene, 2023-10-18:
I was sussessful to compile the 6.6.0-rc5 kernel (link in previous post) including ZFS support without errors, but I'm facing a very basic problem: how to replace the existing kernel with the new one I'd compiled. I tried "make install", but it didn't work. I'm running Experimental Mainline Linux.

Joshua-Riek, 2023-10-18:
Try to build as a deb package with
```
make CROSS_COMPILE=aarch64-linux-gnu- ARCH=arm64 -j "$(nproc)" bindeb-pkg
```

simonswine, 2024-01-21:
I have hit this problem (on a non-dkms OS), it helped patching this symbol restriction in the 5.10 kernel:

https://github.com/armbian/linux-rockchip/pull/142

Joshua-Riek, 2024-02-13:
[image]
I have ZFS working with the latest 6.1 Rockchip BSP kernel.

Joshua-Riek, 2024-03-03:
For anyone wondering, I hope to have Ubuntu 24.04 released in about one months time, and it will contain the previously mentioned 6.1 kernel with zfs.

Joshua-Riek, 2024-05-07:
As ZFS is working on Ubuntu 24.04 I finally had some time to get ZFS working properly on Ubuntu 22.04, this should be in the next release.

cybermeh, 2024-05-25:
I would like to create a RAID1 ZFS pool on Orange Pi 5 Plus. One device would be a M.2 SSD and another would be eMMC. Is it possible to do such thing using ubuntu-rockchip and do you have any tips and tricks?

Joshua-Riek, 2024-05-25:
ZFS will work on ubuntu 22.04 and 24.04 now, so I'm closing this issue as fixed.

closed 2024-05-25 by Joshua-Riek