Joshua-Riek/ubuntu-rockchip #395: Rock 5A with Ubuntu Server v1.24: clock MHz scaling issue?

guynich, 2023-10-02:
Comparing Indiedroid Nova and Rock 5A: both running Ubuntu Server v1.24.  I have a task with heavy CPU load running in the RK3588 `bigcores`.

- I'm seeing the Rock 5A with SOC heatsink attached reach critical temperature after a few minutes with highest `bigcore` temp of 119ºC just before reboot.  The `bigcore` freq changes from 2304 MHz  to **2208** MHz during the run.

- with same task the Indiedroid Nova does not reboot and the highest `bigcore` temperature is 93ºC.  The `bigcore` freq drops from 2256 down to **1608** MHz during the run.

Thus I'm suspicious of the CPU frequency/temperature governance of Rock 5A based on the observations.  Any suggestions for a fix?  Is this something that can be manually configured in a file somewhere?

Joshua-Riek, 2023-10-02:
Use proper cooling, and this will not be an issue.

Joshua-Riek, 2023-10-02:
Governers are to preformance each boot, if you want to disable use:
```
systemctl enable cpu-governor-performance
systemctl enable gpu-governor-performance
```

guynich, 2023-10-03:
I’ll check these commands - thanks Joshua:::

guynich, 2023-10-03:
Confirming I'm able to change the governer on Rock 5A to the different states, e.g.:
- conservative (example below)
- powersave
- ondemand
- performance
- schedutil
```
sudo -i
# See policy to see how CPUs are grouped.
echo conservative > /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor
echo conservative > /sys/devices/system/cpu/cpu4/cpufreq/scaling_governor
echo conservative > /sys/devices/system/cpu/cpu6/cpufreq/scaling_governor
exit

cat /sys/devices/system/cpu/cpu*/cpufreq/scaling_governor 
```
Apart from `powersave` which sets the core clocks to 408MHz, none of these governors addresses the throttling issue: the core temperature of the Rock 5A rises to high levels and the clock frequencies do not get lowered to something like 1200MHz to control the thermal runaway.

I'm thinking this issue is to do with Rock 5A firmware and not under our control here.  Thanks.

closed 2023-10-03 by guynich

guynich, 2023-10-04:
I checked Radxa [releases](https://github.com/radxa-build/rock-5a/releases) specifically rock-5a_ubuntu_jammy_cli_b18.img.xz](https://github.com/radxa-build/rock-5a/releases/download/b18/rock-5a_ubuntu_jammy_cli_b18.img.xz) with Rock 5A and found CPU throttling works as expected with `performance` governor.  This resolves my issue.   Thanks!