Joshua-Riek/ubuntu-rockchip #398: System freeze after few hours

unaiur, 2023-10-05:
This morning, I updated my Ubuntu kernel to 5.10.160-14. In less than 20 hours, the system froze twice, having to reboot it manually.

No errors shown in logs (journalctl).

How can I rollback to previous version?

That version did never freeze.

Joshua-Riek, 2023-10-05:
Can you provide any information on what was going on when the system froze? There where some slight changes to the NanoPC T6 device tree but it should not cause any freezing.

You can download the previous version here:
https://launchpad.net/~jjriek/+archive/ubuntu/rockchip/+build/26699470

To prevent kernel updates:
```
sudo apt-mark hold linux-image-5.10.160-rockchip linux-headers-5.10.160-rockchip
```

unaiur, 2023-10-05:
I do not have any HDMI device connected, just Ethernet and some USB devices.

It is running Home Assistant, but was not doing anything special.

Tomorrow morning I will test just replacing the device tree with the previous one.

I will try also to revert the device tree changes one by one to try to isolate the problematic one.

Joshua-Riek, 2023-10-05:
Cool, thanks, I don't have time to do much testing right now. But I will take a look for sure over the weekend. Maybe dmesg may have some helpful information.

perfectsyrup, 2023-10-05:
@Joshua-Riek I may confirm that desktop version after last update halts after reboot with black screen on NanoPC-T6.
Had the make fresh install on nvme of https://github.com/Joshua-Riek/ubuntu-rockchip/releases/download/v1.27/ubuntu-22.04.3-preinstalled-desktop-arm64-nanopc-t6.img.xz - it works fine. If agree to update all software at first boot, ubuntu reboot in black screen and halts.

unaiur, 2023-10-07:
I do not have any HDMI connected. In my case, the machine freezes after one or two hours.

Joshua-Riek, 2023-10-07:
I narrowed the issue down to a specific commit, that I will revert.

anmansky, 2023-10-08:
last build https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/6373493097 write on eMMC, nanopc-t6 not boot

Joshua-Riek, 2023-10-08:
Yeah, I'm waiting for the fix to be merged, then I'll bump the kernel to the latest kernel versionn from Rockchip and include this fix.

anmansky, 2023-10-08:
Ok

Joshua-Riek, 2023-10-08:
The fix has been merged, and I'm starting a kernel build now on my Launchpad PPA. It should be done in about 4 hours.

Joshua-Riek, 2023-10-08:
The kernel update has been published, and this issue is now fixed.

closed 2023-10-08 by Joshua-Riek