Joshua-Riek/ubuntu-rockchip #400: nvmf-autoconnect.service  fails to start

xaja34, 2023-10-06:
Orange PI 5 plus
Latest Ubuntu server 1.27
I'm booting to NVME drive
Everything seems fine, but when I puul up Cockpit I see the nvmf-autoconnect.service failed to start

Here is the output of 'systemctl status nvmf-autoconnect.service'

× nvmf-autoconnect.service - Connect NVMe-oF subsystems automatically during boot
     Loaded: loaded (/lib/systemd/system/nvmf-autoconnect.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Fri 2023-10-06 17:59:27 UTC; 44min ago

Oct 06 17:59:27 orangepi systemd[1]: Starting Connect NVMe-oF subsystems automatically during boot...
Oct 06 17:59:27 orangepi modprobe[746]: modprobe: FATAL: Module nvme-fabrics not found in directory /lib/modules/5.10.160-rockchip
Oct 06 17:59:27 orangepi systemd[1]: nvmf-autoconnect.service: Control process exited, code=exited, status=1/FAILURE
Oct 06 17:59:27 orangepi systemd[1]: nvmf-autoconnect.service: Failed with result 'exit-code'.
Oct 06 17:59:27 orangepi systemd[1]: Failed to start Connect NVMe-oF subsystems automatically during boot.

Thanks
Bill

Joshua-Riek, 2023-11-03:
The module does not exist with Rockchip Linux 5.10.160.

closed 2023-11-03 by Joshua-Riek

Ukhupacha, 2024-01-13:
Hello Joshua,

Should we just disable the service?