Joshua-Riek/ubuntu-rockchip #406: Mainline experimental CSF Panfrost

StuartIanNaylor, 2023-10-12:
https://www.collabora.com/news-and-blog/news-and-events/pancsf-a-new-drm-driver-for-mali-csf-based-gpus.html#qcom2891

> A [second version of the kernel driver](https://lore.kernel.org/dri-devel/20230809165330.2451699-1-boris.brezillon@collabora.com/) has been posted a couple month back and we are about to post a v3. Development branches can be found [here](https://gitlab.freedesktop.org/panfrost/linux/-/tree/panthor?ref_type=heads) and [here](https://gitlab.freedesktop.org/panfrost/mesa/-/tree/v10+panthor?ref_type=heads) if you want to play with it. They are updated on a regular basis, but I doubt orange pi 5 official images are shipping these drivers.

Joshua-Riek, 2023-10-13:
I saw this, and looks interesting. I've been working on the Turing RK1 recently so have not had time for mainline kernel things. But  I will probably check it out when I find the time.

closed 2023-10-13 by Joshua-Riek

StuartIanNaylor, 2023-10-13:
The HDMI is still a thing but with Arm & Collabora supposedly collaborating maybe the G610 is one of the 1st to benefit but comparing against Panfork should be interesting.
v3 might be good as the feedback might be useful.

magicwenli, 2024-03-04:
News Here: https://www.phoronix.com/news/Arm-Mali-Panthor-DRM-Coming
Mesa PR: https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/26358

Can't wait to give it a try.

StuartIanNaylor, 2024-03-04:
Its should be interesting as that should be an almost complete mainline version, dunno if it will be backported to the BSP or not

StuartIanNaylor, 2024-03-11:
Is there a new Mainline image build on the way?
https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/?h=next-20240308&id=23ed255e74317b10b41b396d4af5b79b47b28c22

ethanc8, 2024-07-20:
The Panthor kernel driver was released to mainline Linux 6.10. Panthor plus Panfrost was [certified OpenGLES 3.1 compliant](https://www.collabora.com/news-and-blog/news-and-events/taming-the-panthor-opengl-es-31-conformance-achived-mali-g610.html) in May, [on kernel 6.10.0-rc1 and Mesa 24.1.1](https://www.khronos.org/conformance/adopters/conformant-products/opengles#submission_1053). Might still not be mature enough to include into these images, though. Armbian edge has linux-6.10 images.

Joshua-Riek, 2024-07-20:
I have no plans on implementing Panthor on Ubuntu 24.04, but i'm likely going to start working on Ubuntu 24.10 with mainline soon.

ethanc8, 2024-07-20:
I see. Nice!

ethanc8, 2024-07-20:
Do you know much about the main differences between Panthor+Panfrost and the rockchip kernel driver + Panfork? Also, is the main point of Panfork to get Panfrost working with the rockchip kernel drivers?

Joshua-Riek, 2024-07-20:
Your borderline spamming..

ethanc8, 2024-07-20:
sorry, I just had a lot of questions. If I have these kinds of questions should I put them in discussions instead?

KylinDemons, 2024-07-20:
> Do you know much about the main differences between Panthor+Panfrost and the rockchip kernel driver + Panfork? Also, is the main point of Panfork to get Panfrost working with the rockchip kernel drivers?

Now I just know that if use panthor instead of panfrost, the mali blob will be unable to use, then like opencl something just won't work. So I think it's not worth to use panthor now.

ethanc8, 2024-07-20:
> Now I just know that if use panthor instead of panfrost, the mali blob will be unable to use, then like opencl something just won't work. So I think it's not worth to use panthor now.

I think Mesa Rusticl might be useful for that case. However, the Arm blob might be better off for OpenCL at least for now, but the best way forward is probably to contribute to Rusticl.

> Panthor the CFS side is in the 6.10 kernel and there is no need to implement anything as it should be just vanilla mainline?

Well, SoCs include more than some CPU cores and some GPU cores wrapped in a bundle. They also have hardware that handles signals for HDMI, USB, Bluetooth, Wi-Fi, and other I/O systems, along with accessing memory. I'm not sure exactly how well mainline Linux handles those, but we know for sure that there are many things that will need to be implemented to support those kinds of peripherals, and that mainline Linux doesn't currently run very well on RK3588.