Joshua-Riek/ubuntu-rockchip #407: build error

tdwornicki, 2023-10-12:
My issue is similar to #344 

./build.sh --board=nanopc-t6 --server-only --mainline --clean --docker

Processing triggers for dbus (1.12.20-2ubuntu4) ...
Repository: 'deb https://ppa.launchpadcontent.net/jjriek/rockchip/ubuntu/ jammy main'
More info: https://launchpad.net/~jjriek/+archive/ubuntu/rockchip
Adding repository.
Adding deb entry to /etc/apt/sources.list.d/jjriek-ubuntu-rockchip-jammy.list
Adding disabled deb-src entry to /etc/apt/sources.list.d/jjriek-ubuntu-rockchip-jammy.list
Adding key to /etc/apt/trusted.gpg.d/jjriek-ubuntu-rockchip.gpg with fingerprint 3CC0D9D1F3F0354B50D24F51F02122ECF25FB4D7
Hit:1 http://ports.ubuntu.com/ubuntu-ports jammy InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports jammy-updates InRelease
Hit:3 http://ports.ubuntu.com/ubuntu-ports jammy-backports InRelease
Hit:4 http://ports.ubuntu.com/ubuntu-ports jammy-security InRelease
Get:5 https://ppa.launchpadcontent.net/jjriek/rockchip/ubuntu jammy InRelease [18.1 kB]
Get:6 https://ppa.launchpadcontent.net/jjriek/rockchip/ubuntu jammy/main arm64 Packages [4776 B]
Get:7 https://ppa.launchpadcontent.net/jjriek/rockchip/ubuntu jammy/main Translation-en [1772 B]
Fetched 24.6 kB in 13s (1966 B/s)
Reading package lists... Done
/bin/bash: line 13: conditional binary operator expected
Error: in ./scripts/build-rootfs.sh on line 111
Error: in ./build.sh on line 156

thx

Joshua-Riek, 2023-10-12:
Weird, I just ran a build with no errors. Ill try on my local system.

https://github.com/Joshua-Riek/ubuntu-rockchip/actions/runs/6496164685

m33ts4k0z, 2023-10-12:
Hello.

I am experiencing the same error when building with docker directly on the orange pi 5 plus.

Joshua-Riek, 2023-10-12:
I found the issue, I forgot to pass a few env vars to docker. Pushing a fix now.

m33ts4k0z, 2023-10-12:
Thanks for the quick fix. I am still experiencing issues when building the image:
```
Setting up swapspace version 1, size = 2 GiB (2147479552 bytes)
no label, UUID=6b391e4b-fd86-4994-b2a7-02d4c2b5e3a8
Created symlink /etc/systemd/system/basic.target.wants/resize-filesystem.service → /lib/systemd/system/resize-filesystem.service.
dpkg-deb: error: '/tmp/camera_engine_rkaiq_rk3588_1.0.3_arm64.deb' is not a Debian format archive
dpkg: error processing archive /tmp/camera_engine_rkaiq_rk3588_1.0.3_arm64.deb (--install):
 dpkg-deb --control subprocess returned error exit status 2
Errors were encountered while processing:
 /tmp/camera_engine_rkaiq_rk3588_1.0.3_arm64.deb
Error: in ./scripts/build-rootfs.sh on line 248
Error: in ./build.sh on line 156
```

Joshua-Riek, 2023-10-12:
You are missing some required packages; please try downloading the following packages.
```bash
sudo apt-get update
sudo apt-get install -y build-essential gcc-aarch64-linux-gnu bison \
qemu-user-static qemu-system-arm qemu-efi u-boot-tools binfmt-support \
debootstrap flex libssl-dev bc rsync kmod cpio xz-utils fakeroot parted \
udev dosfstools uuid-runtime git-lfs device-tree-compiler python2 python3 \
python-is-python3 fdisk bc debhelper python3-pyelftools python3-setuptools \
python3-distutils python3-pkg-resources swig libfdt-dev libpython3-dev
```

m33ts4k0z, 2023-10-12:
I fired up a virtual box VM using Ubuntu 22.04, installed the packages you suggested but I still receive an error using:
`./build.sh --board=orangepi-5-plus -do -m`

```
make[7]: *** [scripts/Makefile.lib:419: arch/arm64/boot/dts/rockchip/px30-engicam-px30-core-ctouch2.dtb] Error 1

make[7]: *** [scripts/Makefile.lib:419: arch/arm64/boot/dts/rockchip/rk3308-roc-cc.dtb] Error 1

make[6]: *** [scripts/Makefile.build:480: arch/arm64/boot/dts/rockchip] Error 2

make[5]: *** [Makefile:1391: dtbs] Error 2

make[5]: *** Waiting for unfinished jobs....

make[4]: *** [Makefile:359: __build_one_by_one] Error 2

make[3]: *** [debian/rules:25: build-arch] Error 2

dpkg-buildpackage: error: make -f debian/rules binary subprocess returned exit status 2

make[2]: *** [scripts/Makefile.package:146: bindeb-pkg] Error 2

make[1]: *** [/media/shared_image_dir/ubuntu-rockchip/build/linux/Makefile:1538: bindeb-pkg] Error 2

make: *** [Makefile:234: __sub-make] Error 2

Error: in ./scripts/build-kernel.sh on line 27

Error: in ./build.sh on line 146
```

Joshua-Riek, 2023-10-12:
I use Ubuntu 22.04 machines for GitHub actions and don't have a problem. So, I'm not sure what is going on here.

https://github.com/Joshua-Riek/ubuntu-rockchip/actions

m33ts4k0z, 2023-10-12:
Any idea why its trying to build assets for other rockchip chips like the px30?

Is that needed?

Joshua-Riek, 2023-10-12:
There is no GPU, NPU, or HDMI working in the mainline. So, there is no reason to build a desktop variant for it. 

I do need to improve the build system to catch errors like this, but it will require a large re-write.

m33ts4k0z, 2023-10-12:
Aha ok so it makes sense now. I am building the non-mainline then. Thanks.

One last question. Do the vulkan drivers work for the orange pi 5 plus? I had mixed results with armbian bookworm.

Joshua-Riek, 2023-10-12:
No vulkan does not work on linux.

m33ts4k0z, 2023-10-12:
Actually I got it to work with Armbian bookworm and even managed to boot the flycast emulator and a game with it but the performance wasnt good. I used the mesa-vulkan-drivers package. I then managed somehow to find a driver update but it stopped working. We can certainly look into this.

cheers

m33ts4k0z, 2023-10-12:
And a last update on this bug report: I still wasnt able to compile doesnt matter which combinations of parameters I tried ornot using parameters at all.

xiaotiansf, 2023-10-17:
> You are missing some required packages; please try downloading the following packages.
> 
> ```shell
> sudo apt-get update
> sudo apt-get install -y build-essential gcc-aarch64-linux-gnu bison \
> qemu-user-static qemu-system-arm qemu-efi u-boot-tools binfmt-support \
> debootstrap flex libssl-dev bc rsync kmod cpio xz-utils fakeroot parted \
> udev dosfstools uuid-runtime git-lfs device-tree-compiler python2 python3 \
> python-is-python3 fdisk bc debhelper python3-pyelftools python3-setuptools \
> python3-distutils python3-pkg-resources swig libfdt-dev libpython3-dev
> ```

With "sudo ./build.sh --board=orangepi-5b -do -m -d", The following build error is still there. dpkg-deb: error: '/tmp/camera_engine_rkaiq_rk3588_1.0.3_arm64.deb' is not a Debian format archive
dpkg: error processing archive /tmp/camera_engine_rkaiq_rk3588_1.0.3_arm64.deb (--install):

xiaotiansf, 2023-10-17:
@Joshua-Riek I guess that "-so" option can be built successfully because camera_engine is not enabled?

Joshua-Riek, 2023-10-17:
Download the package `git-lfs` and try again, use the below commands to checkout lfs objects.
```
git lfs fetch
git lfs checkout
```

xiaotiansf, 2023-10-17:
> Download the package `git-lfs` and try again, use the below commands to checkout lfs objects.
> 
> ```
> git lfs fetch
> git lfs checkout
> ```

Yes. This fix the above error. Thanks. 
But now I got the following errors:
E: Unable to locate package gstreamer1.0-rockchip1
E: Couldn't find any package by glob 'gstreamer1.0-rockchip1'
E: Couldn't find any package by regex 'gstreamer1.0-rockchip1'
E: Unable to locate package mali-g610-firmware
E: Unable to locate package malirun
E: Unable to locate package rockchip-multimedia-config
E: Unable to locate package librist4
E: Unable to locate package librist-dev
E: Unable to locate package rist-tools
E: Unable to locate package libv4l-rkmpp
E: Unable to locate package librockchip-mpp1
E: Unable to locate package librockchip-mpp-dev
E: Unable to locate package librockchip-vpu0
E: Unable to locate package rockchip-mpp-demos
E: Unable to locate package librga2
E: Unable to locate package librga-dev
E: Unable to locate package libwidevinecdm
Error: in ./scripts/build-rootfs.sh on line 255
Error: in ./scripts/build-rootfs.sh on line 255
Error: in ./build.sh on line 156

Joshua-Riek, 2023-10-17:
Delete your working dir and clone the repo again. I have builds running with out any issues.

xiaotiansf, 2023-10-18:
> Delete your working dir and clone the repo again. I have builds running with out any issues.

1. Delete entire cloned dir of repo, 
2. git clone git@github.com:Joshua-Riek/ubuntu-rockchip.git && ubuntu-rockchip
3. git lfs fetch && git lfs checkout
4. sudo ./build.sh --board=orangepi-5b -do -m
I got the exact the same errors: 
E: Unable to locate package gstreamer1.0-rockchip1
E: Couldn't find any package by glob 'gstreamer1.0-rockchip1'
E: Couldn't find any package by regex 'gstreamer1.0-rockchip1'
E: Unable to locate package mali-g610-firmware
E: Unable to locate package malirun
E: Unable to locate package rockchip-multimedia-config
E: Unable to locate package librist4
E: Unable to locate package librist-dev
E: Unable to locate package rist-tools
E: Unable to locate package libv4l-rkmpp
E: Unable to locate package librockchip-mpp1
E: Unable to locate package librockchip-mpp-dev
E: Unable to locate package librockchip-vpu0
E: Unable to locate package rockchip-mpp-demos
E: Unable to locate package librga2
E: Unable to locate package librga-dev
E: Unable to locate package libwidevinecdm
Error: in ./scripts/build-rootfs.sh on line 255
Error: in ./scripts/build-rootfs.sh on line 255
Error: in ./build.sh on line 156

Joshua-Riek, 2023-10-18:
Mainline linux does not have HDMI output, so building a desktop image makes no sense.

xiaotiansf, 2023-10-18:
@Joshua-Riek So with the support of HDMI output, only kernel version 5.10 is available for rockchip?

Joshua-Riek, 2023-10-18:
Yes, there is a 6.1 kernel from Rockchip that should be released in 1-2 months. I'm watching it closely.

m33ts4k0z, 2023-10-18:
> Yes, there is a 6.1 kernel from Rockchip that should be released in 1-2 months. I'm watching it closely.

This is interesting. Do you know if there are any plans of using mainline in the future? Or more correctly said, why do we need to use special kernel versions for the rockchip products?

m33ts4k0z, 2023-10-18:
These 3 parts that are not working are maybe the most important ones :D . I am not familiar with the Linux pipeline but how does this work? Are there volunteers that are slowly adapting the vendor drivers into the kernel by using open source code? 

Is Rockchip giving them code/assisting them into a mainline integration or is Rockchip just doesnt care and leave them figure everything out by themselves?

Joshua-Riek, 2023-10-18:
I don't think people understand, just because HDMI gets patched in. That does not mean you can run a desktop.

xiaotiansf, 2023-10-18:
Mainly interested in rk3588 embedded system. There is no need to have desktop. But HDMI, GPU (and NPU) are critical.

Joshua-Riek, 2023-10-18:
This is why users should use the hacked 5.10 Android kernel. It will be a long time before mainline kernel is fully working for rk3588.

m33ts4k0z, 2023-10-18:
Yeah we don't really have a choice anyway 😁

closed 2023-10-18 by Joshua-Riek