Joshua-Riek/ubuntu-rockchip #408: Custom username and password

windskyxb, 2023-10-13:
How do I customize the username and password when I compile the system myself?I've tried modifying overlay/boot/firmware/user-data, but doesn't seem to work.

Joshua-Riek, 2023-10-13:
I noticed something weird with my latest build about username and password. I don't know if related, but I will take a look tomorrow afternoon.

windskyxb, 2023-10-13:
> I think the issue is that `dsmode: local` is missing in the `meta-data` file.
> 
> ```
> # This is the meta-data configuration file for cloud-init. Please refer to the
> # cloud-init documentation for more information:
> #
> # https://cloudinit.readthedocs.io/
> 
> # Set the datasource mode to "local". This ensures that user-data is acted upon
> # prior to bringing up the network (because everything about the datasource is
> # assumed to be local). If you wish to use an HTTP datasource instead, you can
> # change this to "net" or override it on the kernel cmdline (see README).
> dsmode: local
> 
> # Specifies the "unique" identifier of the instance. Typically in cloud-init
> # this is generated by the owning cloud and is actually unique (to some
> # degree). Here our data-source is local, so this is just a fixed string.
> # Warning: changing this will cause cloud-init to assume it is running on a
> # "new" instance, and to go through first time setup again (the value is
> # compared to a cached copy).
> instance_id: cloud-image
> ```

After I added dsmode: local to meta-date, the recompiled system still cannot log in using the username and password I customized in user-date.I don't know if this is my problem.

Joshua-Riek, 2023-10-13:
I can test in about 2 hours to see what is going on.

windskyxb, 2023-10-14:
I added dsmode: local to meta-data
[image]
Then set the password for root and the username I want to use in user-data to 1234
[image]
But after the compilation is completed, I cannot log in to the system using the user name I set. Only the root user logs in normally. And there is only the user folder of the user ubuntu in the /home directory, and there is no user folder for the user name I set. It seems that the user name I set has not been added to the system.
[image]

Joshua-Riek, 2023-10-14:
Okay, the could-init documentation is awful. However, this is not a problem with my project. The syntax is different when creating users other than "ubuntu" and requires a hashed password.

Example of making a user with name `kim` and password `possible`.
```
users:
  - default
  - name: kim
    passwd: "$6$kW4vfBM9kGgq4hr$TFtHW7.3jOECR9UCBuw9NrdSMJETzSVoNQGcVv2y.RqRUzWDEtYhYRkGvIpB6ml1fh/fZEVIgKbSXI9L1B6xF."
    shell: /bin/bash
    lock-passwd: false
    ssh_pwauth: True
    chpasswd: { expire: False }
    sudo: ALL=(ALL) NOPASSWD:ALL
    groups: users, admin
```
Source: https://stackoverflow.com/questions/61591885/how-do-i-set-a-custom-password-with-cloud-init-on-ubuntu-20-04

A hashed password can be made with:
```
$ mkpasswd -m SHA-512
Password:  <enter your password here>
$6$nq4v1BtHB8bg$Oc2TouXN1KZu7F406ELRUATiwXwyhC4YhkeSRD2z/I.a8tTnOokDeXt3K4mY8tHgW6n0l/S8EU0O7wIzo.7iw1
```

closed 2023-10-15 by Joshua-Riek