Joshua-Riek/ubuntu-rockchip #409: First installation wizzard is crashing

m33ts4k0z, 2023-10-13:
When booting the system for the first time, the wizard is crashing after it configures the locales and while it configures the keyboard. A white window shows saying that the issue will be send to the tracker, a bug reports generates and another window shows showing:

`The installer encountered and unrecoverable error. A desktop session will now be run so that you may investigate the problem or try installing again`

However, after that I receive another wizard and that one works fine and the system boots without a problem. This is image was built from the commit `92771fd` for the Orange Pi 5 Plus.

Joshua-Riek, 2023-10-13:
Did you modify anything? How exactly did you invoke the build script?

m33ts4k0z, 2023-10-13:
I just used this command:  `./build.sh --board=orangepi-5-plus -do`

Joshua-Riek, 2023-10-13:
Thanks for the info, i should be able to take a look this weekend.

m33ts4k0z, 2023-10-14:
Some more context in case it matters:

I used the rockchip tool to flash the image into the emmc and I chose Swedish as the keyboard layout 😊

m33ts4k0z, 2023-10-14:
Well I dont know what happened but I recreated the image with the --clean parameter and this time the error doesnt occur. So I will close this.

closed 2023-10-14 by m33ts4k0z

Joshua-Riek, 2023-10-14:
Ah i see, ill be sure to double-check that it will not crash on my end.