Joshua-Riek/ubuntu-rockchip #411: OPi5Plus: Document how to recover on Linux after installing to SPI

Ferenc-, 2023-10-15:
Hi,

I have followed the [NVMe install instructions here](https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Orange-Pi-5-Plus#install-ubuntu-onto-your-nvme-from-linux). Which was working for a couple of hours, and after that the board stopped booting, and now in a Maskrom state, according to `rkdeveloptool`:

```
./rkdeveloptool ld
DevNo=1	Vid=0x2207,Pid=0x350b,LocationID=303	Maskrom

```

and now it is unclear how to recover the SPI.
On some forums I have read people claiming that `rkdeveloptool db rk3588_spl_loader_v1.08.111.bin` solved the issue,
but for me that is just hangs after it prints `Downloading bootloader...` and never returns.
Stracing the same things shows this where it hangs:

```
ioctl(9, USBDEVFS_SUBMITURB, 0x559798d7e060) = 0
read(5, "\1\0\0\0\0\0\0\0", 8)          = 8
poll([{fd=5, events=POLLIN}, {fd=6, events=POLLIN}, {fd=9, events=POLLOUT}], 3, 60000

```

Any other image that I have tried with `rkdeveloptool` ends up like this.

Ferenc-, 2023-10-17:
> Have you tried with their official instructions on how to do it? They can be found [here](https://drive.google.com/file/d/1enrNSQLKco86w7mUOBl6wU4exos6OMD1/view) Chapter 2.6, page 49

Thanks for showing interest in this issue. I have been through the official instructions a few times.
On page 49 (Chapter 2.6) the instructions assume, that the board is capable of booting from SD card, but in this scenario the board is not booting from an SD card. It is in a permanent Maskrom state.

m33ts4k0z, 2023-10-17:
@Ferenc-  Actually the exact same thing just happend to me 😁

Although I use EMMC, the solution is the same: Erase the SPI.

In the document I linked, it's on page 123, chapter 2.14

Joshua-Riek, 2023-10-17:
Look at the documentation provided by the vendor, I will not document this process.

closed 2023-10-17 by Joshua-Riek

Ferenc-, 2023-10-18:
> @Ferenc- Actually the exact same thing just happend to me 😁
> 
> Although I use EMMC, the solution is the same: Erase the SPI.
> 
> In the document I linked, it's on page 123, chapter 2.14
> 
> After that just restart and it should boot fine.

@m33ts4k0z how do you do that on Linux without Windows? I have tried that tool in Bottles (Wine) but for me that was completely unusable.

m33ts4k0z, 2023-10-18:
@Ferenc- 

Honestly I don't think this is doable on linux since you also need to install Windows drivers and such. 

I haven't asked Shenzhen Xunlong though so I don't know if this can be done on linux. Try reaching out to them on either ali express or on Skype (zhaoyifan_steven).

I got answers to my questions through both channels.

But I the fastest would be to use a Windows computer. I did it on Windows 11 and it took about 3 minutes total if you have the tools setup once.

Ferenc-, 2023-10-21:
> not even with an SD card,

That is the case, not even with an SD card, and the current suggestion would only work once booted.

> will need a windows computer

I have tried Windows 11 in a VM. The tool reports a bunch of errors in Chinese.
And does not really flash what it should, eventually the board remains in the Maskrom state.
I will attempt to get support from OPi.

m33ts4k0z, 2023-10-21:
You can change the language of the tool by editing the the relevant language entry in the ini file in the same directory as the exe. Its totally translated to English. 

I run it on Windows 11 too and I never had any problems surprisingly enough. Have you followed the instructions to the letter? ( Installed drivers, pointed to correct bin files etc)

Try changing the language and post the errors here so we can look into it.