Joshua-Riek/ubuntu-rockchip #417: Fan strategy problem

yuta-sakata, 2023-10-19:
I am a Rock 5B user, and I think the current fan strategy is very poor.  When I watch YouTube videos or other workloads that do not require CPU performance but require GPU performance, the fan spins very slowly or even doesn't spin, which makes my board extremely hot.  Is there a way to permanently change the default strategy to full speed?

yuta-sakata, 2023-10-24:
So, what command should I enter to control my fan?

Joshua-Riek, 2023-10-24:
What tempatures are you getting? The fan should spin up just fine to cool the system. There are thermal trip points defined in the device tree.

yuta-sakata, 2023-10-25:
Oh...Okay, I just checked the device tree of this board and it does not define a fan strategy.  I just defined one myself and the problem was resolved.  
```
fan0: pwm-fan {
		compatible = "pwm-fan";
		#cooling-cells = <2>;
		cooling-levels = <72 94 117 139 162 184 207 229 255>;
		pwms = <&pwm1 0 10000 0>;
		rockchip,temp-trips = <
                        40000   1
                        45000   2
                        50000   3
                        55000   4
                        60000   5
						65000	6
						70000	7
						75000	8
                >;
	};
```

closed 2023-10-28 by Joshua-Riek