Joshua-Riek/ubuntu-rockchip #418: Cant apt update & upgrade

viewh0, 2023-10-20:
Hi, im completly green to ubuntu but i have this error while running sudo apt-get update && sudo apt-get upgrade

Could you tell me what should i do? 

Thanks!

`E: Failed to fetch https://ppa.launchpadcontent.net/jjriek/panfork-mesa/ubuntu/dists/jammy/InRelease  403  Forbidden [IP: 185.125.190.52 443]
E: The repository 'https://ppa.launchpadcontent.net/jjriek/panfork-mesa/ubuntu jammy InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Failed to fetch https://ppa.launchpadcontent.net/jjriek/rockchip-multimedia/ubuntu/dists/jammy/InRelease  403  Forbidden [IP: 185.125.190.52 443]
E: The repository 'https://ppa.launchpadcontent.net/jjriek/rockchip-multimedia/ubuntu jammy InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: Target Packages (main/binary-arm64/Packages) is configured multiple times in /etc/apt/sources.list:3 and /etc/apt/sources.list:43
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:3 and /etc/apt/sources.list:43
W: Target Translations (main/i18n/Translation-en_US) is configured multiple times in /etc/apt/sources.list:3 and /etc/apt/sources.list:43
W: Target Translations (main/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:3 and /etc/apt/sources.list:43
W: Target DEP-11 (main/dep11/Components-arm64.yml) is configured multiple times in /etc/apt/sources.list:3 and /etc/apt/sources.list:43
W: Target DEP-11 (main/dep11/Components-all.yml) is configured multiple times in /etc/apt/sources.list:3 and /etc/apt/sources.list:43
W: Target DEP-11-icons-small (main/dep11/icons-48x48.tar) is configured multiple times in /etc/apt/sources.list:3 and /etc/apt/sources.list:43
W: Target DEP-11-icons (main/dep11/icons-64x64.tar) is configured multiple times in /etc/apt/sources.list:3 and /etc/apt/sources.list:43
W: Target DEP-11-icons-hidpi (main/dep11/icons-64x64@2.tar) is configured multiple times in /etc/apt/sources.list:3 and /etc/apt/sources.list:43
W: Target Packages (universe/binary-arm64/Packages) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list:43
W: Target Packages (universe/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list:43
W: Target Translations (universe/i18n/Translation-en_US) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list:43
W: Target Translations (universe/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list:43
W: Target DEP-11 (universe/dep11/Components-arm64.yml) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list:43
W: Target DEP-11 (universe/dep11/Components-all.yml) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list:43
W: Target DEP-11-icons-small (universe/dep11/icons-48x48.tar) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list:43
W: Target DEP-11-icons (universe/dep11/icons-64x64.tar) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list:43
W: Target DEP-11-icons-hidpi (universe/dep11/icons-64x64@2.tar) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list:43
W: Target Packages (multiverse/binary-arm64/Packages) is configured multiple times in /etc/apt/sources.list:24 and /etc/apt/sources.list:43
W: Target Packages (multiverse/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:24 and /etc/apt/sources.list:43
W: Target Translations (multiverse/i18n/Translation-en_US) is configured multiple times in /etc/apt/sources.list:24 and /etc/apt/sources.list:43
W: Target Translations (multiverse/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:24 and /etc/apt/sources.list:43
W: Target DEP-11 (multiverse/dep11/Components-arm64.yml) is configured multiple times in /etc/apt/sources.list:24 and /etc/apt/sources.list:43
W: Target DEP-11 (multiverse/dep11/Components-all.yml) is configured multiple times in /etc/apt/sources.list:24 and /etc/apt/sources.list:43
W: Target DEP-11-icons-small (multiverse/dep11/icons-48x48.tar) is configured multiple times in /etc/apt/sources.list:24 and /etc/apt/sources.list:43
W: Target DEP-11-icons (multiverse/dep11/icons-64x64.tar) is configured multiple times in /etc/apt/sources.list:24 and /etc/apt/sources.list:43
W: Target DEP-11-icons-hidpi (multiverse/dep11/icons-64x64@2.tar) is configured multiple times in /etc/apt/sources.list:24 and /etc/apt/sources.list:43
W: Target Packages (main/binary-arm64/Packages) is configured multiple times in /etc/apt/sources.list:3 and /etc/apt/sources.list:43
W: Target Packages (main/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:3 and /etc/apt/sources.list:43
W: Target Translations (main/i18n/Translation-en_US) is configured multiple times in /etc/apt/sources.list:3 and /etc/apt/sources.list:43
W: Target Translations (main/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:3 and /etc/apt/sources.list:43
W: Target DEP-11 (main/dep11/Components-arm64.yml) is configured multiple times in /etc/apt/sources.list:3 and /etc/apt/sources.list:43
W: Target DEP-11 (main/dep11/Components-all.yml) is configured multiple times in /etc/apt/sources.list:3 and /etc/apt/sources.list:43
W: Target DEP-11-icons-small (main/dep11/icons-48x48.tar) is configured multiple times in /etc/apt/sources.list:3 and /etc/apt/sources.list:43
W: Target DEP-11-icons (main/dep11/icons-64x64.tar) is configured multiple times in /etc/apt/sources.list:3 and /etc/apt/sources.list:43
W: Target DEP-11-icons-hidpi (main/dep11/icons-64x64@2.tar) is configured multiple times in /etc/apt/sources.list:3 and /etc/apt/sources.list:43
W: Target Packages (universe/binary-arm64/Packages) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list:43
W: Target Packages (universe/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list:43
W: Target Translations (universe/i18n/Translation-en_US) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list:43
W: Target Translations (universe/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list:43
W: Target DEP-11 (universe/dep11/Components-arm64.yml) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list:43
W: Target DEP-11 (universe/dep11/Components-all.yml) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list:43
W: Target DEP-11-icons-small (universe/dep11/icons-48x48.tar) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list:43
W: Target DEP-11-icons (universe/dep11/icons-64x64.tar) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list:43
W: Target DEP-11-icons-hidpi (universe/dep11/icons-64x64@2.tar) is configured multiple times in /etc/apt/sources.list:14 and /etc/apt/sources.list:43
W: Target Packages (multiverse/binary-arm64/Packages) is configured multiple times in /etc/apt/sources.list:24 and /etc/apt/sources.list:43
W: Target Packages (multiverse/binary-all/Packages) is configured multiple times in /etc/apt/sources.list:24 and /etc/apt/sources.list:43
W: Target Translations (multiverse/i18n/Translation-en_US) is configured multiple times in /etc/apt/sources.list:24 and /etc/apt/sources.list:43
W: Target Translations (multiverse/i18n/Translation-en) is configured multiple times in /etc/apt/sources.list:24 and /etc/apt/sources.list:43
W: Target DEP-11 (multiverse/dep11/Components-arm64.yml) is configured multiple times in /etc/apt/sources.list:24 and /etc/apt/sources.list:43
W: Target DEP-11 (multiverse/dep11/Components-all.yml) is configured multiple times in /etc/apt/sources.list:24 and /etc/apt/sources.list:43
W: Target DEP-11-icons-small (multiverse/dep11/icons-48x48.tar) is configured multiple times in /etc/apt/sources.list:24 and /etc/apt/sources.list:43
W: Target DEP-11-icons (multiverse/dep11/icons-64x64.tar) is configured multiple times in /etc/apt/sources.list:24 and /etc/apt/sources.list:43
W: Target DEP-11-icons-hidpi (multiverse/dep11/icons-64x64@2.tar) is configured multiple times in /etc/apt/sources.list:24 and /etc/apt/sources.list:43
`

Joshua-Riek, 2023-10-20:
You are on a very old version that is unfortunately no longer supported or working correctly. You will need to reinstall with the latest available image.

closed 2023-10-20 by Joshua-Riek

jimpil, 2023-10-21:
Hi, I'm having the same issue, and want to make sure I fully understand..are you saying that the only way to fix this is to repeat the image installation, which I think would delete all data? there is no way to upgrade and keep all settings & data? Many thanks in advance...