Joshua-Riek/ubuntu-rockchip #422: Bluetooth Devices Not Detected with rtl8852be on Kernel 5.10.160-rockchip

BitYoungjae, 2023-10-22:
Hello,

I am currently using an Orange Pi 5 plus with the `rtl8852be` ethernet card and facing an issue with Bluetooth device detection. Below are the details:

- **Ethernet Card:** `rtl8852be`
- **Driver Version:** `v1.15.10.0.5-0-gfa2af07cf.20220`
- **Kernel Version:** `5.10.160-rockchip`

**Issue:**
When I utilize the `bluetoothctl` tool and run the `scan on` command, no Bluetooth devices appear in the listing, even when they are in pairing mode.

**Steps to Reproduce:**
1. Turn on a Bluetooth device and set it to pairing mode.
2. On the Orange Pi 5, open the terminal and run `bluetoothctl`.
3. Inside `bluetoothctl`, run the `scan on` command.

Expected Result: Bluetooth devices should appear in the listing.
Actual Result: No Bluetooth devices are listed.

Is this a known issue? If not, do you have any insights or suggestions on what might be causing this problem? Any help or guidance would be greatly appreciated.

Thank you in advance for your support.

BitYoungjae, 2023-10-22:
By the way, the Wi-Fi connection works perfectly fine...

Joshua-Riek, 2023-10-23:
What version did you download? I created a hack / workaround in the v1.28 release to address this issue. Since this hack made a systemd service to load and unload the 8852be kernel module, it has not been distributed to previous releases.

https://github.com/Joshua-Riek/ubuntu-rockchip/releases/tag/v1.28

JFLim1, 2023-10-24:
@Joshua-Riek thank you for sharing and maintaining the wonderful Ubuntu 22.04.3 build.

Received OPI 5 Plus with the so called official PCIe wifi/bluetooth module 8852BE yesterday and flash the latest Ubuntu 22.04.3 v1.28 onto a SD Card. 

Facing the same issue.  No bluetooth connectivity available.  Trying to pair a Logitech bluetooth mouse but unsuccessful.  Stuck at "searching device..."

Also purhased a 256GB EMMC modules but do not seems to know how to install/insert it onto the OPI 5 Plus.  If anyone have a guide how to install/insert the EMMC module onto OPI 5 Plus will be much appreciated.

Deiiji, 2023-10-24:
Blacklist usb and intel bt modules, disable bt service, restart (unplug power) restart bt, is it working after that? Is wifi working after that?

Joshua-Riek, 2023-10-24:
My generic 8852BE module works fine, but I received the official Orange Pi 8852BE module moments ago and will do some testing.

Joshua-Riek, 2023-10-24:
I got it working, needs a small kernel change and some extra firmware. I need to clean my work and double-check a few things.

Joshua-Riek, 2023-10-26:
Quick update - just waiting on my kernel PR to be merged. When that is done, ill publish a firmware and kernel update to fix Bluetooth.

InfernalWolf666, 2023-10-26:
Урааааааа

Deiiji, 2023-10-27:
Thank you Mr J.R.

JFLim1, 2023-10-27:
@Joshua-Riek thank you.  Looking forward to the new firmware and kernel so that RTL8852BE Bluetooth and WIFI both works.

Joshua-Riek, 2023-10-28:
The Kernel and Firmware have finished uploading and are now available on Launchpad.

Please update your system with the below commands, then after a reboot, Bluetooth will be working.
```
sudo apt update
sudo apt upgrade -y
```

closed 2023-10-28 by Joshua-Riek

bagong, 2023-10-28:
Fantastic, thank you so much!!!

InfernalWolf666, 2023-10-28:
🥳🥳🥳🥳🥳

JFLim1, 2023-10-29:
Thank you @Joshua-Riek.  Both Bluetooth Mouse and WIFI works now.  Great!