Joshua-Riek/ubuntu-rockchip #423: [Mainline] Device tree compiled without symbols, overlays not working

baybal, 2023-10-22:
Hello,  I found out that uboot doesn't load my overlay. Looking into logs I got:

```
## Executing script at 00500000
Boot script loaded from mmc 1
reading /ubuntuEnv.txt
311 bytes read in 1 ms (303.7 KiB/s)
reading /dtbs/rk3588s-orangepi-5b.dtb
75629 bytes read in 6 ms (12 MiB/s)
reading /dtbs/overlays/orangepi-5-smart-mass-meter.dtbo
2204 bytes read in 3 ms (716.8 KiB/s)
Applying device tree overlay: /dtbs/overlays/orangepi-5-smart-mass-meter.dtbo
failed on fdt_overlay_apply(): FDT_ERR_NOTFOUND
base fdt does did not have a /__symbols__ node
make sure you've compiled with -@
Error applying device tree overlays, restoring original device tree
reading /dtbs/rk3588s-orangepi-5b.dtb
75629 bytes read in 5 ms (14.4 MiB/s)
reading /vmlinuz
32991744 bytes read in 185 ms (170.1 MiB/s)
reading /initrd.img
30924797 bytes read in 191 ms (154.4 MiB/s)
```

It seems the build script invokes dtc without `-@` flag, so overlay is not working

baybal, 2023-10-22:
CONFIG_OF_OVERLAY is certainly enabled

```
CONFIG_MTD_SPI_NOR=y
CONFIG_MTD_SPI_NOR_USE_4K_SECTORS=y
# CONFIG_MTD_SPI_NOR_SWP_DISABLE is not set
CONFIG_MTD_SPI_NOR_SWP_DISABLE_ON_VOLATILE=y
# CONFIG_MTD_SPI_NOR_SWP_KEEP is not set
CONFIG_MTD_UBI=y
CONFIG_MTD_UBI_WL_THRESHOLD=4096
CONFIG_MTD_UBI_BEB_LIMIT=20
# CONFIG_MTD_UBI_FASTMAP is not set
# CONFIG_MTD_UBI_GLUEBI is not set
# CONFIG_MTD_UBI_BLOCK is not set
CONFIG_MTD_HYPERBUS=m
CONFIG_DTC=y
CONFIG_OF=y
# CONFIG_OF_UNITTEST is not set
CONFIG_OF_FLATTREE=y
CONFIG_OF_EARLY_FLATTREE=y
CONFIG_OF_KOBJ=y
CONFIG_OF_DYNAMIC=y
CONFIG_OF_ADDRESS=y
CONFIG_OF_IRQ=y
CONFIG_OF_RESERVED_MEM=y
CONFIG_OF_RESOLVE=y
CONFIG_OF_OVERLAY=y
CONFIG_OF_NUMA=y
# CONFIG_PARPORT is not set
CONFIG_PNP=y
CONFIG_PNP_DEBUG_MESSAGES=y
```

Joshua-Riek, 2023-10-23:
Please don't open issues related to mainline Linux. It's experimental and a major work in progress. If you have a fix and share it ill be happy to apply it. But I wont be looking into this.

closed 2023-10-23 by Joshua-Riek