Joshua-Riek/ubuntu-rockchip #424: Install to eMMC from previous image possible?

jimpil, 2023-10-23:
Hi many thanks for this ubuntu port :+1:  

I installed an image onto my (onboard) eMMC back in May. I'm now facing updating issues, and would like to update to a newer image. I did see [this instruction](https://github.com/Joshua-Riek/ubuntu-rockchip/wiki/Radxa-ROCK-5B#install-ubuntu-onto-your-emmc-from-linux), but this assumes that the device you're writing to, is NOT the same as the device you're currently running, right? So i guess, my question is, is there an update path which avoids a full re-install? Many thanks in advance...

Joshua-Riek, 2023-10-23:
What update issues are you having? Around April / May I needed to re-work how updates were distributed.

jimpil, 2023-10-23:
Well, it's a similar issue with #418 i think - see screenshot below:

[image]

Joshua-Riek, 2023-10-23:
Ah, yes, you are on a very old install that is somewhat broken now. I recommend re-installing completely, as you won't receive any kernel updates. I made some packaging mistakes that caused this problem...

But you can fix the ppa by doing the below:

Modify the file `/etc/apt/preferences.d/panfork-mesa-ppa` to:
```
Package: *
Pin: release o=LP-PPA-liujianfeng1994-panfork-mesa
Pin-Priority: 1001
```

Modify the file `/etc/apt/preferences.d/rockchip-multimedia-ppa` to:
```
Package: *
Pin: release o=LP-PPA-liujianfeng1994-rockchip-multimedia
Pin-Priority: 1001
```

Then issue your `sudo apt update` command and it should work.

jimpil, 2023-10-24:
Well, I did that, it partially improved the ppas, it prompted me to install
mesa updates, but unfortunately after a reboot everything is messed up
(performance is degraded and chromium won't launch)!

I would NOT suggest that anyone does this...I now need to figure out a way
to revert all the stuff I installed after these changes :(

Joshua-Riek, 2023-10-24:
You will need to re install the system using the latest release. There is not much that I can do since you are running a very old and no longer supported release.

jimpil, 2023-10-25:
OK so, I spent the entire evening yesterday taking a backup, and installing the latest image. I'm now back and running - hopefully I won't need to repeat that any time soon :+1:

jimpil, 2023-10-25:
by the way, it's worth mentioning in the README that one can (permanently) change the default output audio device by editing the last line of  `/etc/pulse/default.pa` (after consulting with `pactl list short sinks`)

closed 2023-10-25 by jimpil