Joshua-Riek/ubuntu-rockchip #426: Network in graphical sever

druid22d, 2023-10-26:
when i install server the network works yet the system in graphical version of server does not recognize that the network is working.  i am using an orange pi 5b...when i boot everything works...then it gets to "waiting for network to configure" it waits 2 minutes and fails.

Joshua-Riek, 2023-10-26:
Do you have ethernet connected?

druid22d, 2023-10-26:
Yes. It works. It is connected and working. Yet the graphical portion
doesn't work.

druid22d, 2023-10-26:
I guess the graphical portion doesn't recognize it for some reason. My
router doesn't seem the orange pi ever either. Which is weird cuz it gives
it an address.

druid22d, 2023-10-26:
Also thank you for your quick response.

Joshua-Riek, 2023-10-26:
Weird, it sounds like cloud-init is acting strange. But in the end, the internet still works? It's just that you must wait 2 minutes before you can log in?

druid22d, 2023-10-26:
Yes correct. The graphical portion doesn't seem that the Internet is
working either.

druid22d, 2023-10-26:
When I use workstation I have no problem with the network. Everything works
fine. I may just switch back to workstation. I can do everything I need
with that. I just wanted to make use of the extra included products with
server. I will just have to do a bit more work to get everything I need.
Lol. Jason Lamy

Joshua-Riek, 2023-10-26:
Since networking works without issue, I don't think this is an issue with this project. Potentially something with cloud-init. Additionally, I did some testing and could not reproduce your problem.

druid22d, 2023-10-26:
It could be an issue with my internet provider. Like I said. I'll probably
just switch back to workstation.

druid22d, 2023-10-26:
Thank you again for your quick response and the great help. Jason Lamy

closed 2023-10-28 by Joshua-Riek

druid22d, 2023-10-29:
is there a way to fix this without resorting to using workstation? or maybe it is the image i used?

Joshua-Riek, 2023-10-29:
I could not reproduce this issue; maybe try downloading the latest release. Or try the desktop image, which will require a keyboard and mouse for the first use. However, it does use a different network manager, so the desktop image may solve the problem.

druid22d, 2023-10-29:
I always use a keyboard etc. I haven't used a headless system yet. That is
my next project. I want to build a cluster for distributed computing. I
have always wanted my own "Super Computer ". Lol. Thanks again for your
help. Jason Lamy.

druid22d, 2023-10-29:
I ended up switching to workstation. No problem now. I cannot understand
why it doesn't work in server. I have a regular PC that the same issue
happens with server. I believe that it is my internet provider. There is
some sort of issue with them. Thanks again. Jason Lamy

druid22d, 2023-10-30:
maybe if i set a static IP then maybe it would work...I'll have to
flash server on a card and test it out with a static IP address. Maybe that
would fix the issue. my isp will give the orange pi an ip address but i
cannot see it on the router...very weird....may be good for torrenting....

druid22d, 2023-10-30:
It must be my isp....i just installed ubuntu inside a vm on my windows
machine and got the same problem. i used a native version of ubuntu...so it
must be some issue here. i will have to call them and figure it
out....maybe i need to use a static IP address....