Joshua-Riek/ubuntu-rockchip #427: usb type-c: no power on port after unplugging display for a long time

Serjbory, 2023-10-26:
Hello! 
The issue I've faced with:
I have a display connected to Orange 5 via usb type-C cable If I unplug the cable from the display and then plug it again in a few seconds/minutes - everything is ok - display is detected. But if I left it unplugged for a few hours (and more) -> the OS is not able to detect it again (as any other usb-device connected via this cable).

**Details:**
 
Autosuspend - is off (including inner configs in ` /sys/module/usbcore/parameters/autosuspend is set to "-1"`)

`/sys/bus/usb/devices/$hub-$port/power/control is set to "on"`

**System details:**
Ubuntu 22.04.3 LTS (jammy)
Linux version 5.10.160-rockchip (root@bos01-arm64-014)

**dmesg:**
`[ 2078.323684] usb 7-1: Product: Touchscreen
[ 2078.323688] usb 7-1: Manufacturer: ELAN
[ 2078.346979] input: ELAN Touchscreen as /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.11.auto/usb7/7-1/7-1:1.0/0003:04F3:2AD0.0009/input/input28
[ 2078.347275] input: ELAN Touchscreen UNKNOWN as /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.11.auto/usb7/7-1/7-1:1.0/0003:04F3:2AD0.0009/input/input29
[ 2078.347447] input: ELAN Touchscreen UNKNOWN as /devices/platform/usbdrd3_0/fc000000.usb/xhci-hcd.11.auto/usb7/7-1/7-1:1.0/0003:04F3:2AD0.0009/input/input30
[ 2078.347886] hid-multitouch 0003:04F3:2AD0.0009: input,hiddev96,hidraw3: USB HID v1.10 Device [ELAN Touchscreen] on usb-xhci-hcd.11.auto-1/input0
[ 2079.311415] rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_enable] Update mode to 1920x1080p60, type: 10(if:200, flag:0x0) for vp2 dclk: 148500000
[ 2079.311638] rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_enable] dclk_out2 div: 2 dclk_core2 div: 2
[ 2079.311649] rockchip-vop2 fdd90000.vop: [drm:vop2_crtc_atomic_enable] set dclk_vop2 to 148500000, get 148500000
[ 2079.342006] dw-dp fde50000.dp: full-training link: 4 lanes at 5400 MHz
[ 2079.352032] dw-dp fde50000.dp: clock recovery succeeded
[ 2079.360725] dw-dp fde50000.dp: channel equalization succeeded`

**lsusb:**
`
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               3.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 
  bDeviceProtocol         3 
  bMaxPacketSize0         9
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0003 3.0 root hub
  bcdDevice            5.10
  iManufacturer           3 Linux 5.10.160-rockchip xhci-hcd
  iProduct                2 xHCI Host Controller
  iSerial                 1 xhci-hcd.8.auto
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x001f
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
[... 552 lines cut ...]
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 006: ID 04f3:2ad0 Elan Microelectronics Corp. Touchscreen
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 002: ID 17ef:6099 Lenovo Lenovo Traditional USB Keyboard
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 002: ID 17ef:608d Lenovo Optical Mouse
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`

**xrandr:**
`Screen 0: minimum 16 x 16, current 1920 x 1080, maximum 32767 x 32767
XWAYLAND0 connected primary 1920x1080+0+0 (normal left inverted right x axis y axis) 290mm x 170mm`

**NOTE**: The same display and Orange 5 works fine with Armbian (without this issue)

Serjbory, 2023-11-29:
Same bug with 1.30 build

Serjbory, 2023-11-29:
Same problem with 1.30 build

Serjbory, 2023-11-29:
Same problem with build 1.30

AMWKey, 2024-02-07:
I have encountered exactly the same issue! As for latest build 1.33 can confirm that problem is still there. I did some tests with previous builds and noticed that such behaviour first appeared on v.1.15 and seems like all later versions inheritated it, earlier builds (1.14 for example) works fine without this issue

AMWKey, 2024-12-10:
I think I managed to find a workaround. 

I faced the same issue on orangepi5 at the latest build which is v.2.4.0 now and found out that when you connect type-c display, port0-partner appears in /sys/class/typec/ along with port0 directory already present there and if you unplug display, port0-partner directory in /sys/class/typec/ disappears with only port0 remaining. So I did some tests with plugging display and when unplugging it and leaving it unplugged for 8-10 hours, suddenly in that window "fake" port0-partner directory appeared and kept typec port busy preventing real display from connecting. I tested it with old v1.14 build and fake port0-partner never appeared there. 

So the fix to wake up display was to find the driver name in /sys/class/typec/port0/device/driver (in my case it was 6-0022) and execute commands: 
echo 6-0022 > /sys/bus/i2c/drivers/typec_fusb302/unbind
echo 6-0022 > /sys/bus/i2c/drivers/typec_fusb302/bind

Serjbory, 2024-12-13:
Hello ! Thank you @AMWKey for this workaround.

Is there any chance to re-open this issue and fix the bug in further versions?

Serjbory, 2024-12-13:
Is there any chance to re-open this issue and fix the bug in further versions?

Serjbory, 2024-12-13:
Reopen please